innoveit/react-native-ble-manager #134: ios BleManager.connect Exception

panmin, 2017-04-22:
i used(run ios device):
BleManager.connect('EA:EC:6B:1B:5D:98')
            .then((peripheralInfo) => {
                // Success code
                console.log('Connected');
            })
            .catch((error) => {
                // Failure code
                console.log('error');
            });
but it don't go to  Success code and  Failure code，it has an error:
ExceptionsManager.js:71 Exception '*** -[__NSPlaceholderArray initWithObjects:count:]: attempt to insert nil object from objects[0]' was thrown while invoking connect on target BleManager with params (
    "EA:EC:6B:1B:5D:98",
    23
)

please help me!!

catherinessssss, 2017-04-22:
Same problem.

catherinessssss, 2017-04-22:
Same Problem. 

let peripheralId = this.state.ble.id;
BleManager.connect(peripheralId).then((peripheralInfo) => {
	console.log('Connected');
	console.log(peripheralInfo);
})
.catch((error) => {
        console.log('error');
});

It don't go to Success code and Failure code.

marcosinigaglia, 2017-04-22:
Versions?

catherinessssss, 2017-04-22:
My version is 3.1.2

catherinessssss, 2017-04-22:
It doesn't  work after I've updated the version to 3.2.0.

panmin, 2017-04-24:
@catherinessssss did you run ios?

catherinessssss, 2017-04-24:
Yes.

panmin, 2017-04-24:
@catherinessssss  “this.state.ble.id” the value is?

catherinessssss, 2017-04-24:
"this.state.ble.id" is the id of my sensor device. Have you solve this problem?

panmin, 2017-04-24:
@catherinessssss in android the id is "EA:EC:6B:1B:5D:98"，but in ios the id is not the value,the same device.

catherinessssss, 2017-04-24:
I have not run my program on android before. But I can get the id when it running on ios. 
Have you import the BleManager.xcodeporj to your ios libraries?

panmin, 2017-04-24:
@catherinessssss yes,i had import it to my ios libraries with xcode，i use 3.2.0 . what's you error info

catherinessssss, 2017-04-25:
There was not error show up. It's just do go success code and failure code.

panmin, 2017-04-26:
@marcosinigaglia i write the mac address wrong,when i write it right,it works success.can you solve the bug? when the mac address error,it can go to failure code.

marcosinigaglia, 2017-04-26:
Fix in 3.2.1

closed 2017-04-26 by marcosinigaglia