innoveit/react-native-ble-manager #137: How can I get serviceUUID of an already connected bluetooth device?

sahil290791, 2017-05-01:
The `BleManager.getConnectedPeripherals([])` returns empty array even though the device is connected.

marcosinigaglia, 2017-05-02:
Hi, OS, versions?

sahil290791, 2017-05-02:
Hi, I am testing it on Android 7.1.2. Basically I need to read battery level of my bluetooth headset when connected with my phone.

marcosinigaglia, 2017-05-02:
Hi, now the method works only with peripheral that you already scanned.
Try to edit the method like this:
```
WritableArray map = Arguments.createArray();
BluetoothManager manager = (BluetoothManager) context.getSystemService(Context.BLUETOOTH_SERVICE);
List<BluetoothDevice> devices = manager.getConnectedDevices(BluetoothProfile.GATT);
for(BluetoothDevice device : devices) {
	if(device.getType() == BluetoothDevice.DEVICE_TYPE_LE) {
		Peripheral peripheral = new Peripheral(device, reactContext);
		Boolean accept = false;

		if (serviceUUIDs != null && serviceUUIDs.size() > 0) {
			for (int i = 0; i < serviceUUIDs.size(); i++) {
				accept = peripheral.hasService(UUIDHelper.uuidFromString(serviceUUIDs.getString(i)));
				}
			} else {
				accept = true;
			}

			if (peripheral.isConnected() && accept) {
				try {
					Bundle bundle = BundleJSONConverter.convertToBundle(peripheral.asJSONObject());
					WritableMap jsonBundle = Arguments.fromBundle(bundle);
					map.pushMap(jsonBundle);
				} catch (JSONException ignored) {
					callback.invoke("Peripheral json conversion error", null);
				}
			}
		}
}
```

Let me know if it's working.

sahil290791, 2017-05-02:
Thanks @marcosinigaglia  will let you know by end of the day.

marcosinigaglia, 2017-05-03:
Hi @sahil290791 , do you have any updates?

sahil290791, 2017-05-03:
@marcosinigaglia sorry got busy with someother work, will try it out in next few hours and share the update.

sahil290791, 2017-05-03:
@marcosinigaglia Hi, can you tell me where should I add that code? Its not JS which I can add in react native component. And how do I get UUID of any device?

marcosinigaglia, 2017-05-04:
You have to change `getConnectedPeripherals` in the Android part (Java).
To retrieve the UUID you have to do a scan or maybe with getConnectedPeripherals after the changes.

marcosinigaglia, 2017-05-09:
Hi, have you tried?

marwakhalil, 2017-05-10:
hi, i tried the basic example and i couldn't get the list of serviceUUIDs after scan,
i'm developing on a mac and test on android 6
thanks

yefengong, 2017-05-18:
BleManager.java
```java
import android.bluetooth.BluetoothProfile;

..............

	@ReactMethod
	public void getConnectedPeripherals(ReadableArray serviceUUIDs, Callback callback) {
		Log.d(LOG_TAG, "Get connected peripherals");
		WritableArray map = Arguments.createArray();
		BluetoothManager manager = (BluetoothManager) context.getSystemService(Context.BLUETOOTH_SERVICE);
		List<BluetoothDevice> devices = manager.getConnectedDevices(BluetoothProfile.GATT);
		for(BluetoothDevice device : devices) {
			if(device.getType() == BluetoothDevice.DEVICE_TYPE_LE) {
				Peripheral peripheral = new Peripheral(device, reactContext);
				Boolean accept = false;
				if (serviceUUIDs != null && serviceUUIDs.size() > 0) {
					for (int i = 0; i < serviceUUIDs.size(); i++) {
						accept = peripheral.hasService(UUIDHelper.uuidFromString(serviceUUIDs.getString(i)));
					}
				} else {
					accept = true;
				}

				if (peripheral.isConnected() && accept) {
					try {
						Bundle bundle = BundleJSONConverter.convertToBundle(peripheral.asJSONObject());
						WritableMap jsonBundle = Arguments.fromBundle(bundle);
						map.pushMap(jsonBundle);
					} catch (JSONException ignored) {
						callback.invoke("Peripheral json conversion error", null);
					}
				}

			}

		}
		callback.invoke(null, map);
	}
```

in react native code:
```js
    BleManager.getConnectedPeripherals([])
      .then((peripheralsArray) => {
        // Success code
        console.warn('Connected peripherals: ' + JSON.stringify(peripheralsArray));
      }).catch((err) => {
        console.warn('aaaa')
    });
```

result is:` Connected peripherals:[]`

Did I do something wrong?

marcosinigaglia, 2017-05-19:
Hi @yefengong , the code appear correct, are you sure the device is a BLE?
Try to remove the `if(device.getType() == BluetoothDevice.DEVICE_TYPE_LE) ` check and look if you find something different.

yefengong, 2017-05-19:
@marcosinigaglia the device is a BLE, `peripheral.isConnected() ` return false.
I tried `getConnectedDevices`, but return empty.
Now I try to `getPairedPeripherals` first, and then try to connect in a loop.
I'd like to do the connect method in the native module side, do you have any idea on the code? 
```java
	@ReactMethod
	public void getConnectedPeripherals(ReadableArray serviceUUIDs, Callback callback) {
		Log.d(LOG_TAG, "Get paired peripherals");
		WritableArray map = Arguments.createArray();
		BluetoothManager manager = (BluetoothManager) context.getSystemService(Context.BLUETOOTH_SERVICE);
		BluetoothAdapter mBtAdapter = manager.getAdapter();
		Set<BluetoothDevice> pairedDevices = mBtAdapter.getBondedDevices();

		if(!pairedDevices.isEmpty()) {
			for(BluetoothDevice device : pairedDevices) {

				Peripheral peripheral = new Peripheral(device, reactContext);
				Boolean accept = false;
				if (serviceUUIDs != null && serviceUUIDs.size() > 0) {
					for (int i = 0; i < serviceUUIDs.size(); i++) {
						accept = peripheral.hasService(UUIDHelper.uuidFromString(serviceUUIDs.getString(i)));
					}
				} else {
					accept = true;
				}
				
				if (accept) {
	
					if(!peripheral.isConnected()) {
						Log.d(LOG_TAG, "!peripheral.isConnected()");
					}
					try {
						Bundle bundle = BundleJSONConverter.convertToBundle(peripheral.asJSONObject());
						WritableMap jsonBundle = Arguments.fromBundle(bundle);
						map.pushMap(jsonBundle);
					} catch (JSONException ignored) {
						callback.invoke("Peripheral json conversion error", null);
					}
				}
				
			}
		}

		callback.invoke(null, map);
	}
```

marcosinigaglia, 2017-05-23:
You can add a new method to retrieve the paired peripherals and then connect from the RN.

yefengong, 2017-05-23:
@marcosinigaglia Thanks, I did it.

closed 2017-05-25 by marcosinigaglia