innoveit/react-native-ble-manager #138: Lack of bluetooth information.

ZuzooVn, 2017-05-05:
Hi guys.

I am using react-native-ble-manager v3.2.1, I want to get data from connected device with below code:

```
BleManager.connect('XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX')
  .then((peripheralInfo) => {
    // Success code
    console.log('Connected');
    console.log(peripheralInfo);
  })
  .catch((error) => {
    // Failure code
    console.log(error);
  });
```

My problems are the peripheralInfo does not hold enough Bluetooth data information. 
I connected my Bluetooth device with Android native app, everything is ok. I can receive whole Bluetooth data.

So, how can I get full Bluetooth data from react-native-ble-manager?

Thanks,

x5c, 2017-05-05:
sorry OOT, did you success to show the list devices ? i didnt work..help

marcosinigaglia, 2017-05-05:
Hi @ZuzooVn , what data is missing?

x5c, 2017-05-05:
Hi Master... @marcosinigaglia can you help me ?

marcosinigaglia, 2017-05-05:
@x5c please open an issue and add more informations.

ZuzooVn, 2017-05-05:
Hi @marcosinigaglia 
For example, I sent 100byte via Bluetooth, but the peripheralInfo get only 70 byte

marcosinigaglia, 2017-05-05:
I don't understand, you read the data from a characteristic?

ZuzooVn, 2017-05-08:
AFAIK, `peripheralInfo` variable holds the Bluetooth information, right?
My problem was that peripheralInfo keep only 2/3 data that has been seen via Bluetooth, so I receive the corrupt data.

marcosinigaglia, 2017-05-08:
I don't know what's your situation, if your Bluetooth is continuously send data, like a heartbeat monitor, i think you write your value in a characteristic and you read this value from your app. If your data is not costant you have to subscribe to a characteristic from the the app.

closed 2017-05-15 by marcosinigaglia