innoveit/react-native-ble-manager #139: not getting data from ble module to react app

arunajayathilaka, 2017-05-07:
I am using react 16.0.0 alpha 6 and react native 0.44.0 and ble-manager version ^3.2.1.
ble module successfully connected to ble react app. And also show all the services and characteristics.
I send data from ardiuno serial monitor to app. I use read API of ble manager library to read data. API is working but getting string is not I entered in serial monitor. And also same string appear all the time when clicking on read button in app.

 ```
read(){
      BleManager.read('A8:1B:6A:88:EA:DE','ffe0','ffe1')
      .then((readData) => {
        // Success code
        alert(readData);
        console.log('Read: ' + readData);
      })
      .catch((error) => {
        // Failure code
        console.log(error);
      });

    }

```
and also 
```
NativeAppEventEmitter
            .addListener('BleManagerDidUpdateValueForCharacteristic', this.read );
```
when I enter something on serial monitor It will trigger and but getting same string (A81B6A88EADE)

marcosinigaglia, 2017-05-08:
Hi, maybe you have to use `startNotification` instead the `read` method.

arunajayathilaka, 2017-05-08:
yeah it works for me.. Can I know what is the purpose of having read API??

arunajayathilaka, 2017-05-08:
yeah it works for me. Can I know What is the purpose of having read API?

marcosinigaglia, 2017-05-08:
When the device change continuously the value, like for reading some measure, the app need only to update this value sometimes. In this case you usually user `read` method.
I think you can choose in Arduino this approach but I haven't experience.

closed 2017-05-08 by marcosinigaglia