innoveit/react-native-ble-manager #141: [Android] Getting 'invalid peripheral UUID' error when connecting using MAC address

jussisaurio, 2017-05-09:
Hi,

I can successfully scan for BLE devices and get them to return their name & ID (mac address), but I cannot connect to any device because I always get "Invalid peripheral UUID" from the connect method. According to the readme I should be able to connect with MAC as well, is this correct?

**### Version**
- react-native-ble-manager v3.2.1
- react-native v0.44.0

**### Expected behaviour**

Should connect to discovered peripheral based on MAC address

**### Actual behaviour**

get 'Invalid peripheral uuid'

**### console log**

05-09 14:20:49.099 10255 10134 I ReactNativeJS: [ { rssi: -70,
05-09 14:20:49.099 10255 10134 I ReactNativeJS:     name: 'foo',
05-09 14:20:49.099 10255 10134 I ReactNativeJS:     id: 'B8:27:EB:EF:2D:22',
05-09 14:20:49.099 10255 10134 I ReactNativeJS:     advertising:
05-09 14:20:49.099 10255 10134 I ReactNativeJS:      { CDVType: 'ArrayBuffer',
05-09 14:20:49.099 10255 10134 I ReactNativeJS:        data: 'AgEGEQbCZwNjF44wtg9ArpAh/FINEwhycGktYWlycXVhbGl0eS1ib3gAAAAAAAAAAAAAAAAAAAAAAAAAAAA=' } } ]
05-09 14:37:42.938 10255 10134 I ReactNativeJS: 'handleConnectPress', 'B8:27:EB:EF:2D:22'
05-09 14:37:42.956 10255 10134 I ReactNativeJS: 'error connecting to peripheral', 'Invalid peripheral uuid'

marcosinigaglia, 2017-05-09:
Hi @jussisaurio , it's very strange, if you have this error it meas it fail the `BluetoothAdapter.checkBluetoothAddress` method. Are you sure you don't change something in the address? Check if you are passing a simple string.

jussisaurio, 2017-05-09:
@marcosinigaglia Yep, I checked my Java stack trace and the method was getting a null parameter. I had a dumb error in one of my redux functions. Case closed :)

closed 2017-05-09 by jussisaurio