innoveit/react-native-ble-manager #145: Scan List

marwakhalil, 2017-05-12:
I was able to integrate the react native ble manager and i used the simple example 
i didn't have the list of devices after scan
i was able to capte other devices but not the devices around
what should i do to get all the list ?
Thanks

marcosinigaglia, 2017-05-12:
Hi, versions? What do you mean with "other devices"?

marwakhalil, 2017-05-12:
those are versions : 
 "base64-js": "^1.2.0",
    "react": "16.0.0-alpha.6",
    "react-native": "0.44.0",
    "react-native-ble-manager": "^3.2.1",
i'm developing in a mac os 10.12
example: i got three phones around enabling their bluetooth and visible and when i launch the app i keep on detecting a device from the other room

marwakhalil, 2017-05-12:
All i want is to get a list of connected devices around and click on a one for sending data 
but scanning doesn't give that list

marcosinigaglia, 2017-05-12:
Are the BLE devices advertising?

marwakhalil, 2017-05-12:
what do you mean by advertising ? i'm sorry i didn't understand the meaning

marcosinigaglia, 2017-05-12:
This library is only for connect as client to a BLE device, you need a "server" app in the other phones.

closed 2017-05-12 by marcosinigaglia

marwakhalil, 2017-05-12:
ah ok thanks

marwakhalil, 2017-05-12:
i'm actually intending to use a react native application and connect it to a getaway, for sending data to it, i'm totally new with react native and i really wanna know if this library allows me to do this feature ?

marcosinigaglia, 2017-05-12:
Try this https://github.com/jacobrosenthal/react-native-ble

marwakhalil, 2017-05-12:
Okay, thanks