innoveit/react-native-ble-manager #146: Question about Write in hex

stephenjen, 2017-05-17:
The write function expects data to be in Base64 format, however, my BLE module expects all writes to it to be in hex.  I'm currently not getting a response when I write to my module and I'm wondering if I'm sending data in the wrong format.  Is there a way to get this to work with data in hex?

Thanks!

closed 2017-05-17 by stephenjen

Philipwzj, 2017-06-19:
How did you solve this?

stephenjen, 2017-06-22:
This worked for me on version 5.0.3:

`var url = "WHAT_YOU_WANT_TO_WRITE";`
`            var data1 = [];`
`            for (var i = 0; i < url.length; i++){`
`                data1.push(url.charCodeAt(i));`
`            }`

Then

`BleManager.write(id, service, characteristic, data1)`
`.then(() => {`
`              })`