innoveit/react-native-ble-manager #147: Send data

marwakhalil, 2017-05-18:
Hello,
Does this library allow me to connect and sending data to DA14680 chip (a ble device) through my react native app ? 
Thanks

marcosinigaglia, 2017-05-19:
Hi @marwakhalil , if the chip support the BLE you can connect and write/read data.

closed 2017-05-19 by marcosinigaglia

marwakhalil, 2017-05-19:
Hi @marcosinigaglia   Thanks, and how about connecting to an advertised smartphone, i'm actually advertising my android phone for testing as an iBeacon i couldn't connect, will that be possible cause i've noticed that the mac id is generated, so will it be possible to connect to a smartphone advertised as beacon

marcosinigaglia, 2017-05-19:
No, you can only act as a client, if you have experience you can implement the advertsing part.

marwakhalil, 2017-05-19:
A okay thanks for that man