innoveit/react-native-ble-manager #149: Android not triggering BleManagerDiscoverPeripheral

stephenjen, 2017-05-19:
I'm just following the sample code and so far been successful at getting the iOS app to connect.  Not having success on the Android side.

Here's my code
componentDidMount() {

        BleManager.enableBluetooth()
        .then(() => {
                  // Success code
                  console.log('The bluetooth is already enabled or the user confirm');
        })
        .catch((error) => {
                  // Failure code
                  console.log('The user refuse to enable bluetooth');
        });  
        BleManager.start({showAlert: false});
        this.handleDiscoverPeripheral = this.handleDiscoverPeripheral.bind(this);
        NativeAppEventEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral );
        }
        .catch((error) => {
                // Failure code
                console.log(error);
        });
}
handleDiscoverPeripheral(data){
        console.log('Got ble data', data);
        this.setState({ ble: data })
}

Console log shows the success code for BleManager.enableBluetooth, The bluetooth is already enabled or the user confirm, but the handleDiscoverPeripheral callback for BleManagerDiscoverPeripheral listener is never triggered.  

Here's what I have in AndroidManifest.xml:

    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>

    <uses-sdk
        android:minSdkVersion="21"
        android:targetSdkVersion="23" />

Any pointers would be greatly appreciated.

marcosinigaglia, 2017-05-19:
Add also:
```
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
```

stephenjen, 2017-05-19:
I completely rolled the Android code back to the example and noticed that android stores the id as 44:A6:E5:04:8E:B5, where as iOS stores it as 6513D5F6-D102-45F9-8884-B1D360AA3C50...  So the connect part code, which wasn't included in my initial snippet, had the wrong id :(

I tried this because I noticed that the handleDiscoverPeripheral wasn't called in iOS as well.

closed 2017-05-19 by marcosinigaglia

vldgeorgiev, 2017-10-31:
@stephenjen It's an old issue, but I just noticed that setting targetSdkVersion to 23 prevents the BleManagerDiscoverPeripheral from firing, just like your problem. Setting it to 22 fixes the issue.
The Bluetooth permissions are set.
I will investigate it some more.

vldgeorgiev, 2017-10-31:
Yeah, with SDK 23, in addition to the permissions in AndroidManifest.xml
```
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"/>
```
I also had to explicitly ask the user for the ACCESS_COARSE_LOCATION permission like
```
  PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION)
    .then(granted => {
      if (granted === 'granted') {
        BleManager.scan(...)
        ...
      }
    });
```