innoveit/react-native-ble-manager #150: Trouble in reading write's response promise

alexandrapersea, 2017-05-22:
Hello,

I'm trying to read write's response. The documentation says that write returns a Promise object. From what i see there's no example on how to read it.

This is my code:
```
BleManager.write('XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX', 'XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX', 'XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX', data)
  .then((resp) => {
    console.log(resp);
  })
  .catch((error) => {
    console.log(error);
  });
```
but resp is undefined.

Sorry if this is a dumb question but what am i doing wrong or how am i supposed to read the response?

marcosinigaglia, 2017-05-22:
Hi, the 'response' is only the confirmation that you write the value. If you expect some in return from the device maybe you have to use the notification.

alexandrapersea, 2017-05-23:
From what i see `startNotification`  returns a fulfilled promise with no value. I'm a bit confused on how to get to the actual response of the device after write.

marcosinigaglia, 2017-05-23:
After you started the notification you retrieve the values from `BleManagerDidUpdateValueForCharacteristic` event.

closed 2017-05-25 by marcosinigaglia