innoveit/react-native-ble-manager #152: undefined is not a function (evaluating 'transform.forEach')

Carole89, 2017-05-23:
### Version
- react-native-ble-manager v 3.3.0
- react-native v 0.44.0
- windows 10 (64 bit)

### package.json

```
{
  "name": "TestApp",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest"
  },
  "dependencies": {
    "react": "16.0.0-alpha.6",
    "react-native": "0.44.0",
    "react-mixin": "^3.0.5",
    "react-timer-mixin": "^0.13.3",
    "react-native-ble-manager": "^3.3.0"
  },
  "devDependencies": {
    "babel-jest": "20.0.3",
    "babel-preset-react-native": "1.9.2",
    "jest": "20.0.3",
    "react-test-renderer": "16.0.0-alpha.6"
  },
  "jest": {
    "preset": "react-native"
  }
}
```

please i need help when i run the ExampleBle in android device i am getting this Error:

> undefined is not a function (evaluating 'transform.forEach') 

thanks

marcosinigaglia, 2017-05-23:
Hi @Carole89 , I'm working on the examample project, take the example from the documentation.

closed 2017-05-23 by marcosinigaglia

Carole89, 2017-05-23:
thank you @marcosinigaglia for your repply.

Yes, i execute the example which is included in README.md.  [Basic Example](https://github.com/innoveit/react-native-ble-manager/blob/master/README.md#basic-example)

But I still have the same error

marcosinigaglia, 2017-05-23:
Check the installation part and clean the project.