innoveit/react-native-ble-manager #154: write() never succeed while notification on

w8choi, 2017-05-27:
### Version
- react-native-ble-manager v3.2.1
- react-native v0.44.0

### Expected behaviour

- BleManager.write(peripheralId, serviceUUID, characteristicUUID, data) should successfully write to BLE device while receiving notification regularly at PRIORITY_HIGH (interval=12)

### Actual behaviour

- Throwing "Write failed" every time I try to write something to the device. Tried solution #103 but didn't solve the problem either.

### Steps to reproduce

1. start notification
2. try writing to ble device

### Stack trace and console log

marcosinigaglia, 2017-05-29:
Hi, iOS or Android? Have you tried to add an higher timeout delay?

w8choi, 2017-05-29:
on both actually.. I've tried adding 1- 5000 with no success :(

w8choi, 2017-05-29:
actually I think it's working on iOS if I remove the event listener and write and then add the listener again but that didn't solve it on android

marcosinigaglia, 2017-05-30:
Hi, check on the device documentation, unfortunately each device has some difference and behaviors.

closed 2017-06-08 by marcosinigaglia