innoveit/react-native-ble-manager #156: Trying to read in subscription callback fails

vikeri, 2017-05-30:
### Version

Tell us which versions you are using:
- react-native-ble-manager: 3.3.0
- react-native: 0.44.0
- Android, Nexus 5X

### Expected behaviour

Doing a read in the callback of a subscription should be succesful

### Actual behaviour

The read always fails, so I have to retry it in the `catch` statement with a timeout, then it works. It seems that the BLE is "busy" and thus is not ready to read yet. I don't know if it is the peripheral or the Android BLE manager that is not ready yet. 

### Steps to reproduce

1. Subscribe to a characteristic
2. In the callback for the subscribe event, read the same characteristic

### Stack trace and console log

It just says, read failed.

marcosinigaglia, 2017-05-30:
Hi @vikeri , try to add some delay after the subscription.

vikeri, 2017-05-30:
@marcosinigaglia Yeah that works, but I don't really see it as a solution. I'm using the library in the background, and then it doesn't work. (setTimeout does not run when the app is in the background). Do you have an idea of why it fails like this?

marcosinigaglia, 2017-05-30:
I don't know why, a lot of devices needs some delay after connection and subscription. You have to add this delay in the native Android code..maybe in the future i can added it as a parameter.

vikeri, 2017-05-30:
@marcosinigaglia Alright, did some research and it seems that this could be a solution maybe? http://www.brendanwhelan.net/2015/bluetooth-command-queuing-for-android
Some sort of command queue for Android.

marcosinigaglia, 2017-05-31:
Hi @vikeri, I'll study the code, in my scenario I add the delay only in the connection and subscription phase, then I send data every 20 millisecond so I don't know if is it the best solution.

vikeri, 2017-05-31:
@marcosinigaglia Yeah I trust you to make the best assessment on how to solve it :)

whitedogg13, 2017-05-31:
Hi @marcosinigaglia  , I might spot a possible cause.

In line 404 @ Peripheral.java, we have codes like this:
```
if (gatt.writeDescriptor(descriptor)) {
   ...
   callback.invoke();
   ...
}
```
This call flow uses `writeDescriptor` in a synchronous manner. However, from Android doc https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html, we can see the `writeDescriptor` just returns whether the operation is `initialied` and the actual result comes from `onDescriptorWrite` callback.

I will try to fix this and report it later :)

marcosinigaglia, 2017-05-31:
@whitedogg13 yes, I think is this the problem.

whitedogg13, 2017-06-01:
Hi @marcosinigaglia @vikeri , I just tested the solution and it works(TI Sensor Tag + Android 7).
Please refer to the [PR](https://github.com/innoveit/react-native-ble-manager/pull/158), thanks :)

closed 2017-06-01 by marcosinigaglia

vikeri, 2017-06-01:
@whitedogg13 that's great! Well done!