innoveit/react-native-ble-manager #159: Restarting BLE device does not reconnect it to app

stephenjen, 2017-06-01:
When my app and BLE device are turned on in this sequence:
1. Turn app on
2. Turn BLE device on
The app and device connects and I'm able to get data to and from the BLE device.

However, if I then turn off the device and turn it back on again, I'm not able to get the device to reconnect.  

The question: how can I get the app and device to reconnect after either on is turned off and then turned back on?

Here's the code I have:

Main component
```
componentWillMount() {
    ....
    ....
    this.props.getRawBLEStream('16AEAAAE-E7FD-4699-A0D6-5BDBEE84D32D');

 }
```

Inside my action creator
```
export const getRawBLEStream = (id) => {

  // console.log('getRawBLEStream', id);
  return (dispatch) => {
    console.log('1');
    BleManager.start({ showAlert: false });
    BleManager.connect(id)
      .then((peripheralInfo) => {
        console.log('4');
        // Success code
        console.log('BLE connected!!');
        startBLENotification(id, dispatch);
        writeToDevice(id);
      })
      .catch((error) => {
        console.log('5');
        // Failure code
        console.log(error);
        // BleManager.disconnect(id);
      });

    if (Platform.OS === 'android' && Platform.Version >= 23) {
      PermissionsAndroid.checkPermission(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION)
        .then((result) => {
          if (result) {
            console.log('Permission is OK');
          } else {
            PermissionsAndroid.requestPermission(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION)
              .then((result) => {
                if (result) {
                  console.log('User accept');
                } else {
                  console.log('User refuse');
                }
              });
          }
        });
    }
    // return (dispatch) => { dispatch({ type: GET_BLE_STREAM }); };
  }
};

const writeToDevice = (id) => {

  const writeToDeviceInterval = this.setInterval(() => {
    const encoded = base64.encode('PH610BADC');
    BleManager.write(id, '1000', '1001', encoded)
      .then(() => {
      })
      .catch((error) => {
        // it comes here when connection is distrupted
        console.log(error);
        clearInterval(writeToDeviceInterval);
        Actions.measure({ refresh: { title: 'Settings', isGettingRawStream: false } });
      });
  }, 1000);
};
```

The writeToDevice function fails when the BLE device is turned off, and I call a react-native-router-flux function to refresh the main component.

marcosinigaglia, 2017-06-01:
Hi, you have to call BleManager.start only one time to init the module, try to scan if the first connect fail.

stephenjen, 2017-06-01:
Yes, if I force it to start and connect again, it will work fine after turning the BLE device off and on.  

I'm just wondering how it is done the first time where the app is on but device is off, then the device is turned on, and magically it connects?  Seems like the app was waiting for the connection, but I have no such listener.

marcosinigaglia, 2017-06-01:
I don't know the tech details but sometime the smartphone already know the device so can connect direcly, probably through the device advertising.

stephenjen, 2017-06-02:
I think I need to use:

`BleManager.removePeripheral(id);`

but it require me to upgrade from version 3.3.0.  On 5.0.3 my write statement no longer works.  The data for the .write() was encoded using the following, and it worked for version 3.3.0:

`const encoded = base64.encode('PH610BADC');`

but now on 5.0.3 I get this error:

`JSON value 'UEg2MTBCQURD' of type NSString cannot be converted to NSArray`

What gives?

stephenjen, 2017-06-02:
Ok, need to create the data this way:

```
                var url = "PH610BADC";
                var data1 = [];
                for (var i = 0; i < url.length; i++){
                    data1.push(url.charCodeAt(i));
                }
```

stephenjen, 2017-06-02:
Here's what I finally did to get the reconnect to work.  Not sure why it is, but `BleManager.connect` seems to wait for a connection to be made, so the goal for me became looking for ways to call `BleManager.connect` after a disconnection.  The natural place seems to be in the callback for the BleManagerDisconnectPeripheral listener, and that worked.

closed 2017-06-02 by stephenjen