innoveit/react-native-ble-manager #161: Scanning for service-uuids not working on Android

vikeri, 2017-06-09:
### Version

Tell us which versions you are using:

- react-native-ble-manager v5.0.3
- react-native v0.44.0
- Android 7.1.2

### Expected behaviour

When scanning only for specific service UUID's it should find all devices with those ids.

### Actual behaviour

It finds no devices at all.

### Steps to reproduce

1. Start scan and specify a service UUID in the array
2. Have an available BLE unit with that service UUID
3. See that nothing is found in the scan

### Stack trace and console log

No errors thrown. I checked the native logs and the UUID is printed correctly [here](https://github.com/innoveit/react-native-ble-manager/blob/b7a3e44656f85e67b23474af5acd5fc9bc63eeb1/android/src/main/java/it/innove/LollipopScanManager.java#L55)

marcosinigaglia, 2017-06-09:
I think the code is wrong.

marcosinigaglia, 2017-06-09:
Can you try this change?
`ScanFilter filter = new ScanFilter.Builder().setServiceUuid(new ParcelUuid(UUIDHelper.uuidFromString(serviceUUIDs.getString(i)))).build();`

marcosinigaglia, 2017-06-09:
There are some issues with the scanning:
https://github.com/iDevicesInc/SweetBlue/wiki/Android-BLE-Issues

vikeri, 2017-06-09:
Changed it and it still doesn't work... Even opened it in the debugger and it the UUID is correct there. Maybe an Android issue as you suggested

vikeri, 2017-06-09:
The SweetBlue library seems pretty sweet (pun intended). Maybe migrate to that as a backend in the future if these issues persist?

marcosinigaglia, 2017-06-12:
Yes, is not a bad idea.

vikeri, 2017-07-03:
This was an internal bug. Someone accidentally stopped broadcasting our services...

closed 2017-07-03 by vikeri