innoveit/react-native-ble-manager #162: serviceUUID, charactersUUID how to get these two parameters

songxiaoliang, 2017-06-11:
I have now connected to the success of the implementation of how to read operation, which is how to read the contents of the inside of the Senate serviceUUID, charactersUUID how to get these two parameters?

marcosinigaglia, 2017-06-12:
You have to check in the device documentation or use the [retrieveServices](https://github.com/innoveit/react-native-ble-manager#retrieveservicesperipheralid) method.

closed 2017-06-12 by marcosinigaglia

songxiaoliang, 2017-06-12:
Yes, I registered. I now do not know how to get the serviceUUID, charactersUUID these two values, and then passed to the read function to read the data operation。Here's my code：

setTimeout(() => {
             
              BleManager.retrieveServices(peripheral.id).then((peripheralData) => {
                  
                  ToastAndroid.show('service' + JSON.stringify(peripheralData), ToastAndroid.SHORT);
                  BleManager.readRSSI(peripheral.id).then((rssi) => {
                      ToastAndroid.show('RSSi'+rssi, ToastAndroid.SHORT);
                  });

              });

          }, 900);

marcosinigaglia, 2017-06-12:
Check in peripheralData for services and characteristics.

songxiaoliang, 2017-06-13:
@marcosinigaglia Hi, I'm in 
BleManager.retrieveServices (peripheral.id) .then ((peripheralData) => {
                       ToastAndroid.show ('Retrieved External Service' + JSON.stringify (peripheralData), ToastAndroid.SHORT);
} 
Get peripheralData, now how to remove serviceUUID and charactersUUID from this peripheralData? Can you give me a BleManager.read example, very grateful, look forward to your reply

marcosinigaglia, 2017-06-13:
Hi, if you log through the console you see easily the json structure, you have two arrays peripheralData.services and peripheralData.characteristics with all the information.

songxiaoliang, 2017-06-13:
@marcosinigaglia Hi, I have consolemed the contents of peripheralData, but I found that there are many arrays in peripheralData. And I do not know how to get the serviceUUID and charactersUUID of the Bluetooth device that the current phone is already connected to. How to get from the peripheralData to the current connection of the Bluetooth device corresponding serviceUUID and charactersUUID? Looking forward for your reply, thank you

marcosinigaglia, 2017-06-14:
You have to check in the logs or in the bluetooth device documentation.