innoveit/react-native-ble-manager #163: How to rewrite the "getConnectedPeripherals" method to get paired peripherals for IOS

yefengong, 2017-06-11:
If a user paired and connected a BLE device at "Settings-> Bluetooth", it return none after the "BleManager.scan" method.

So, It's possible to rewrite the "getConnectedPeripherals" method to get paired peripherals for IOS?

Actually, "getConnectedPeripherals" is better then  "getPairedPeripherals". I tried "getConnectedPeripherals", but return an empty array.

LeoLeBras, 2017-06-11:
I am also facing the same issue.

marcosinigaglia, 2017-06-12:
Hi @yefengong , you can create a new function that use `getPairedPeripherals` and make a PR

whitedogg13, 2017-08-02:
Hi @LeoLeBras @marcosinigaglia ,

Per my understanding, there's `no API to list paired devices for iOS`. SO discussion:
* https://stackoverflow.com/questions/28337159/how-to-list-discoverable-bluetooth-devices-and-already-paired-devices-in-ios-wh/28342831#28342831

Maybe there're some hacky way to do so, but I don't think there will be a correct approach for this library.

Thanks!

closed 2018-01-11 by marcosinigaglia