innoveit/react-native-ble-manager #168: Received advertising data format is not ArrayBuffer

me-shaon, 2017-06-13:
### Version

- react-native-ble-manager v5.0.4
- react-native v0.45.1
- Android v5.0.2

### Expected behavior

From your example code, it is shown that in `handleDiscoverPeripheral`, the `peripheral` object will have a **advertising** key and it will be an **ArrayBuffer** ( the same thing also specified in [cordova-plugin-ble-central documentation](https://github.com/don/cordova-plugin-ble-central#advertising-data) ). it can then be easily converted to UInt8Array or String.

### Actual behavior

When I logged the `peripheral` object it is showing that the `advertising` key is an object with two keys: 'CDVType' and 'data'. I tried to convert this object to UInt8Array using `new Uint8Array(peripheral.advertising)` and to String using `String.fromCharCode.apply(null, new Uint16Array(buf))`. But failed. It is showing an empty result.

### Stack trace and console log
~[Console log](https://image.ibb.co/i10u2k/ble_manager.png)

whitedogg13, 2017-06-13:
Hi, you can try to use this npm module https://www.npmjs.com/package/base64-js.
After npm install, you can convert the adv packet to byte array like this:
```
import * as Base64Js from "base64-js";
...
let bytes = Base64Js.toByteArray(device.advertising.data);
```
However, I do think it's good idea to convert the bytes inside the library itself.

me-shaon, 2017-06-13:
@whitedogg13 thanks for your reply. I just did a similar workaround using [this base-64 plugin](https://www.npmjs.com/package/base-64) by decoding the `device.advertising.data`. As I am new to this type of project, I'm not sure if this is the proper way to do this. Because it wasn't specified in any of the documentations and this issue should be clarified with a proper example, I think.

whitedogg13, 2017-06-14:
@marcosinigaglia How do you think about this idea (using bytes rather than base64 for advertising data) ? I can try to make a PR about this one, for both iOS and Android.

marcosinigaglia, 2017-06-15:
Hi @whitedogg13 , I already converted the other method using byte array but I missed the advertising part, your help is welcome.

guerjon, 2017-06-15:
On LollipopScanner onScanResult i get the advertising data like this

SparseArray<byte[]> array = result.getScanRecord().getManufacturerSpecificData();
byte[] manufacture_data = [] 
if(array.valueAt(0) != null){
      manufacture_data = array.valueAt(0); 
}
and you get an array of bytes with the manufacturerData inside.

whitedogg13, 2017-06-16:
@marcosinigaglia Cool, with these utils the patch is quite straightforward. I have made a [PR](https://github.com/innoveit/react-native-ble-manager/pull/170) for this, after this we should be able to access `bytes` directly, thanks!

closed 2017-06-16 by marcosinigaglia