innoveit/react-native-ble-manager #171: how do i get the local bluetooth address/mac/uuid?

biks152207, 2017-06-19:


whitedogg13, 2017-06-19:
Hi @biks152207 ,

It is in the `id` property if you do a `BleManager.scan`.
This property has different meaning according to platforms:
* For iOS, it doesn't let you get the MAC address of a device, it gives you a random UUID instead.
* For Android, it represents the MAC address obtained by [BluetoothDevice.getAddress](https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#getAddress())

biks152207, 2017-06-19:
@whitedogg13  but how can i get the local bluetooth address?it gives the address of other device not the address of the device which is running the application.I have been scratching my head for long time.

whitedogg13, 2017-06-19:
Hi @biks152207 ,
As far as I know, there are `public` and `random` addresses for LE devices defined in spec. What do you mean by `local` address? and what's the use case you want to achieve?

biks152207, 2017-06-19:
@whitedogg13 the address of bluetooth of the device which is running the application.which is responsible for scanning other ble devices

biks152207, 2017-06-19:
@whitedogg13  suppose i am scanning other devices ble address from my android device..i want to get my android device ble address, not the remote one.

biks152207, 2017-06-19:
@whitedogg13  i have my application requirement which is, the device which is running the application shares the profile address with its own bluetooth address to the server.and the device in other end try scan the bluetooth address who has shared the profile address (scanning is pretty straight forward).it sends the address and get the profile from the server.The problem here is the device which shares the profile couldn't get its own bluetooth address,probably public address according to you. I hope you understand.

whitedogg13, 2017-06-19:
Hi @biks152207 
After Android M, there's no official way to obtain device's MAC address. You can see this SO [discussion](https://stackoverflow.com/questions/33377982/get-bluetooth-local-mac-address-in-marshmallow).

And just for you reference, regarding your use case BLE has `peripheral`(been connected) mode and `central`(connect to others) mode, generally Android devices only act as `central`. So I think what you described here might be impractical to acommplish using pure apps even beside the get address issue you describe here. 

Hope this helps!

biks152207, 2017-06-19:
@whitedogg13 thanks.but how one device can scan other device bluetooth mac address.if the device can get mac address of other device than why it doesn't get its own mac address.are you trying to say, bluetooth has private and public mac address(android), uuid(ios).

biks152207, 2017-06-19:
@whitedogg13  i think i have to say peripheralId instead of mac address.

whitedogg13, 2017-06-20:
@biks152207 You're welcome. 
Actually you're right about `if the device can get mac address of other device than why it doesn't get its own mac address`, the address is indeed been exchanged during BLE procedure, and it's the OS (Android/iOS) itself blocks your access for privacy reason.

biks152207, 2017-06-20:
@whitedogg13  i have a requirement in my app..the requirement is like. any user who wants to share the contact address, he sends the profile id and his mobile bluetooth address.so other user opens his device and scans the bluetooth address and send those address to the server..and server matches the given mac address who have shared the contact and give those contact as response.

Basically who wants to share contact give his mac address to the server and user in other end scans the nearby devices and send it to the server and server gives all the contacts shared by given mac address.
So what do you think, how can i deal with this situation.

whitedogg13, 2017-06-20:
hi @biks152207 ,
I think your application requires both BLE peripheral and central modes, and when in peripheral mode, you need to put custom advertising packet for others to scan.
As mentioned before, most Android devices only support central mode (however iOS does support both modes), so it will be hard to properly implement such a feature.

In alternative, maybe you can consider using barcode generation / scanning to exchange data, and use some library like node-uuid to get uinique identifier.

BTW, since `BleManager` library actually targets on central mode BLE,  maybe we should discuss this elsewhere :)

closed 2017-06-21 by marcosinigaglia

tmjordan, 2021-06-13:
Hey guys,
Please can you explain me why on an iphone and on android, for the same device we don't have the same device id (whether it be uuid or mac address)

marcosinigaglia, 2021-06-17:
@tmjordan , is a security policy of iOS, you can't use the mac address and the uuid is unique for the device.

tmjordan, 2021-06-19:
@marcosinigaglia thanks for your answer. Can i at least get the unique uuid for my own device (i mean local device not remote)

tmjordan, 2021-06-19:
@marcosinigaglia is there a way to set an id generated by myself. So when i discover nearby devices, i can have that id for each device discovered ?

marcosinigaglia, 2021-07-07:
@tmjordan no