innoveit/react-native-ble-manager #174: Does it work for SSP?

nishiltamboli, 2017-06-21:
First of all, thanks for the amazing BLE library! It works pretty much well with BLE.

I was just curious to know if the same library would work for SPP profile in Bluetooth? In my case, I am trying to connect to speakers on Bluetooth 2.1. I am sure SPP is the profile which I am looking at - Serial Port Profile. Can you confirm if this library should work?

whitedogg13, 2017-06-21:
@nishiltamboli 
Unfortunately, SPP over BLE is not available. 
However, using BLE as `transport` layer to emulate serial port should be pretty straightforward. You can see  more info [here](https://devzone.nordicsemi.com/question/71/is-there-a-serial-port-profile-for-ble/).
BTW, I do think this is an awesome lib, thanks for @marcosinigaglia !

marcosinigaglia, 2017-06-21:
Thanks @whitedogg13 :)

closed 2017-06-22 by marcosinigaglia