innoveit/react-native-ble-manager #176: Connect TimeOut

Philipwzj, 2017-06-22:
Currently I saved a device ID, and automatically connect to that device when the app is open. For Android, the connect function will fire the error in 1 minute, and tell me that the device cannot de connected. However, for iOS, the connect function never give me the error.

` BleManager.connect(global.deviceID)
            .then(() => {
                BleManager.retrieveServices(global.deviceID)
                    .then((peripheralInfo) => {
                        console.log(peripheralInfo);
                    }).catch((error) => {
                    this.setState({visible: false});
                    setTimeout(() => {
                        Alert.alert(
                            'ERROR',
                            'Device cannot be connect',
                            [
                                {
                                    text: 'Reconnect', onPress: () => {
                                    this.setState({visible: true});
                                    this.connectSensor()
                                }
                                },
                                {
                                    text: 'Rescan', onPress: () => {
                                    clearTimeout(timeOut);
                                    this.startScan()
                                }
                                },
                                {text: 'Cancel', onPress: () => console.log('Cancel Pressed'), style: 'cancel'},
                            ],
                            {cancelable: false}
                        );
                    }, 100)
                });
            })
            .catch((error) => {
                this.setState({visible: false});
                setTimeout(() => {
                    Alert.alert(
                        'ERROR',
                        'Device cannot be connect',
                        [
                            {
                                text: 'Reconnect', onPress: () => {
                                this.setState({visible: true});
                                this.connectSensor()
                            }
                            },
                            {
                                text: 'Rescan', onPress: () => {
                                clearTimeout(timeOut);
                                this.startScan()
                            }
                            },
                            {text: 'Cancel', onPress: () => console.log('Cancel Pressed'), style: 'cancel'},
                        ],
                        {cancelable: false}
                    );
                }, 100)
            });`

marcosinigaglia, 2017-06-23:
Hi, it's strange, can you debug on the android code? If you connect the first time the callback cannot be call again to give you the error.

Philipwzj, 2017-07-10:
I think that's an iOS issue, which will allow the device to to connect to the bluetooth device forever. I edited the ObjectiveC code, in order to set the timeout

mateusz-, 2017-08-18:
I did something like this to set a custom timeout:

```js
try {

  await new Promise(async (resolve, reject) => {

    this.connectTimeout = setTimeout(reject, 5000);

    await BleManager.connect(controller.id);

    if(this.connectTimeout) {

      clearTimeout(this.connectTimeout);

      this.connectTimeout = null;

      resolve();

    }

  });

} catch(err) {

  clearTimeout(this.connectTimeout);

  this.connectTimeout = null;

  console.error('Could not connect to device.');

}
```

closed 2017-08-28 by marcosinigaglia

garbo-a, 2018-02-01:
Due to documentation:
https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518766-connect

> Attempts to connect to a peripheral do not time out. To explicitly cancel a pending connection to a peripheral, call the cancelPeripheralConnection(_:) method. 

` this.timeouId = setTimeout(() => {
              BleManager.disconnect(id);
              this.timeouId = null;
            }, 5000);`