innoveit/react-native-ble-manager #178: How To Read Multiple Characteristics At The Same Time

andycamp, 2017-06-26:
### Version

react-native-ble-manager v 5.1.1
react-native v 0.45.1
- iOS/Android - both

I am new to javascript, React Native, and ble. So, please bear with me.

I am trying to read several characteristics at the same time, and I'm not able to.

```
BleManager.retrieveServices(peripheral.id).then((peripheralData) => {
      console.log('Retrieved peripheral services for speed', peripheralData);
      // BleManager.readRSSI(peripheral.id).then((rssi) => {
      //   console.log('Retrieved actual RSSI value', rssi);
      // });
      BleManager.read(peripheral.id, '0000', '0002').then((readData) => {
          // Success code
          console.log('2 = ' + readData);
        })

    }).catch((error) => { console.log(error)})

BleManager.retrieveServices(peripheral.id).then((peripheralData) => {
      console.log('Retrieved peripheral services for speed', peripheralData);
      BleManager.read(peripheral.id, '0000', '0001').then((readData) => {
          console.log('1 = ' + readData);
        })

    }).catch((error) => { console.log(error)})

```

I have tried many variations of this same code, and I only get data for the last characteristic read. Any help in fixing my code would be greatly appreciated.

Thanks.

marcosinigaglia, 2017-06-27:
Hi @andycamp, after the connection you have to call `retrieveServices` only once and than read.
I think you have to wait the first read result before starting another read.

andycamp, 2017-06-28:
@marcosinigaglia - Thank you so much for your response. It helped a lot. I ended up getting my code working. Just to help me clear things up in my head, once I've connected and scanned a ble for services, if I store the peripheral id and serviceUUID that I want to call, I can then read the characteristics as many times as I want?

marcosinigaglia, 2017-06-29:
If you don't close the connection you don't need to call `retrieveServices` again and you can read as many times you want. Then if you connect again to this peripheral you have to call `retrieveServices` again.

andycamp, 2017-06-29:
Great. Thank you.

closed 2017-06-29 by andycamp