innoveit/react-native-ble-manager #179: peripherals not disconnected in background, can't disconnect/discover them

peterhoneder, 2017-06-28:
### Version

- react-native-ble-manager v5.1.0
- react-native v0.45.0
- iOS/Android v7.1.1

### Expected behaviour

It should be possible to disconnect already connected peripherals in all cases, also if the app reloads (through the react native debug menu).

### Actual behaviour

Reloading the app through the react native debug menu will leave a previously connected peripheral connected, and does not allow to either discover or disconnect it.

### Steps to reproduce

1. start BleManager, scan for peripherals, connect one
2. reload react-native UI
3. start BleManager, scan for peripherals

Now the peripheral will not be found by scanning, not be part of getConnectedPeripherals, but still be connected (I can see the device log on my BLE device).

### Side note:

This also seems to happen when the app receives an OutOfMemoryError. Then when you open it again it will have the same symptoms.

peterhoneder, 2017-06-28:
I have tried calling getConnectedDevices(GATT) from BluetoothManager, it returns an empty list.

marcosinigaglia, 2017-07-05:
Hi, if you scan again change something?

aeons, 2017-07-10:
I'm facing something like this as well.

If I scan and connect to a device, then live reload the app, then the peripheral shows up when I call `getConnectedPeripherals`, but I am unable to disconnect from it. That is, calling `disconnect` works fine, but the `BleManagerDisconnectPeripheral` event is never emitted.

adyanced, 2017-08-29:
same.
Any crash while debugging will cause peripheral can't be scanned again because it is still connected. Have to reboot my ble device (I can see it's in the connected status) or take my phone away for a while so peripheral will disconnet.

allenhsu, 2018-01-03:
+1 on Android, after connected to a device, when I refresh the app in dev build, `getConnectedPeripherals` returns nothing but it does not appear in scan list, which means it's actually still connected.

Works fine on iOS

closed 2018-01-11 by marcosinigaglia

ramjak, 2018-04-10:
Is this already closed? I found a work around by detecting a appState change then disconnecting the device. But it just doesn't work on app reload, it won't wait for disconnecting and ignore the rest of code. Anyone have a better solution?
```js
componentDidMount() {
    AppState.addEventListener('change', this.handleAppStateChange);
  }

  handleAppStateChange = (nextAppState) => {
    if (nextAppState.match(/inactive|background/)) {
      console.log('App has come to the inactive!');
      let count = 0;
      if (this.props.connectedBike) {
        while (count < 200) {
          count += 1;
          console.log('didit');
        }
        BleManager.disconnect(this.props.connectedBike.id)
          .then(() => {
            // Success code
            console.log(this.props.connectedBike);
          })
          .catch((error) => {
            // Failure code
            console.log(error);
          });
      }
    }
```

rnevius, 2018-10-25:
Indeed, the peripheral needs to be reconnected. Maybe this has something to do with caching of the connected peripheral(s) before reload? As mentioned, `getConnectedPeripherals()` returns previously-connected peripherals, but they are not actually connected.

GoodnatureBeacon, 2020-04-17:
The problem is in the BleManager.java file. The old BleManager doesn't get released/garbage collected and so maintains it's connections, if you don't have a timeout on your scan all these Managers will keep scanning forever too. I tried fixing but couldn't figure out how to do it. :/ Is there any fix out there?