innoveit/react-native-ble-manager #184: I can't write the message to anoth

Eleven1108, 2017-07-05:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

Eleven1108, 2017-07-05:
when i use the Module of write it always tell me that  Writing is not permitted. why i can't write the message to Bluetooth? thanks

marcosinigaglia, 2017-07-05:
Probably the characteristics is not writable.

closed 2017-07-31 by marcosinigaglia