innoveit/react-native-ble-manager #185: retrieveServices sometimes takes a long time, even after just connecting?

JClackett, 2017-07-07:
### Version

Tell us which versions you are using:

- react-native-ble-manager v5.1.1
- react-native v0.42.3
- iOS/Android iphone6 iOS 10.1

Not exactly an error, but sometimes the promise from the retrieveServices function takes 10+ seconds to be resolved, wondering if anyone else is having this issue?

marcosinigaglia, 2017-07-18:
Sometimes happen also with my divices, only with Android. I use a custom timeout instead waiting the callback to avoid this problem.

closed 2017-08-03 by JClackett

marcosinigaglia, 2017-08-03:
Hi @JClackett , do you find something?

JClackett, 2017-08-03:
I didn't but created timeout too

Tanisi001, 2021-05-24:
Hi,
We tried adding a custom timeout of 15 seconds but still faced this issues.
We can see that in the first attempt, BLEManager.retreiveServices took around 14 seconds to return and our timeout is 15 seconds which led to timeout.
In the subsequent attempt, BLEManager.retreiveServices did not return any response till 15 seconds.

Used:
- react-native-ble-manager v6.6.2
- react-native v0.59.8
- DeviceModel: iPhone12,1, version: 14.6

Is there any solution for this?