innoveit/react-native-ble-manager #190: It takes a while to install, is that ok?

hybrisCole, 2017-07-30:
### Version

Tell us which versions you are using:

- react-native-ble-manager v0.6.1
- react-native v0.46.1
- iOS/Android v.?

### Expected behaviour

I'm guessing yarn install should not take a lot on react-native-ble-manager

### Actual behaviour

yarn install last for about 15minutes installing react-native-ble-manager, is that ok? not sure but if so, I would add it on docs, I stopped like 3 times assuming it got stuck :P

### Steps to reproduce

1. yarn add react-native-ble-manager
2. wait
3. see errors when building, that some files can't be found and stuff like that
4. ???
5. profit

### Stack trace and console log
Albertos-MBP:BLETentRN albertocole$ yarn install
yarn install v0.27.5
[1/4] Resolving packages...
[2/4] Fetching packages...
[3/4] Linking dependencies...
[4/4] Building fresh packages...
[1/1] ⠄ react-native-ble-plx: --- xcodebuild: WARNING: Unable to open project file '/Users/albertocole/Desktop/InternetTent/BLETentRN/BLETentRN/node_modules/react-native-ble-plx/ios/BleCl
[-/1] ⡀ waiting...
[-/1] ⡀ waiting...
[-/1] ⡀ waiting...
^C/1] ⡀ waiting...
Albertos-MBP:BLETentRN albertocole$ rm -rf node_modules/
Albertos-MBP:BLETentRN albertocole$ yarn install
yarn install v0.27.5
[1/4] Resolving packages...
[2/4] Fetching packages...
[3/4] Linking dependencies...
[4/4] Building fresh packages...
success Saved lockfile.
Done in 674.48s.

```
```

marcosinigaglia, 2017-07-31:
Sorry but I don't use yarn, I don't have problem with npm.

whitedogg13, 2018-04-13:
@marcosinigaglia I just setup a new RN project and try `yarn add react-native-ble-manager`, and everything works correctly. Maybe we can close this one if no further update from issuer.

closed 2021-02-18 by marcosinigaglia