innoveit/react-native-ble-manager #192: unhandled promise rejection

Ideasrefined, 2017-08-04:
### Version

Tell us which versions you are using:

- react-native-ble-manager v_latest as of 4thAug2017
- react-native 
react-native-cli: 2.0.1
react-native: 0.47.0

- iOS/Android v.Android 7.1.1

I have linked ble-manager library using react-native link. I followed the instructions to change minSDK version to 18.

I am getting the following Warning when I use 

`"BleManager.start({showAlert: false, allowDuplicates: false});"`

```
Possible Unhandled Promise Rejection (id: 0):
TypeError: Cannot read property 'start' of undefined
TypeError: Cannot read property 'start' of undefined
```
I have traced the problem to the highlighted call.
```
start(options) {
    return new Promise((fulfill, reject) => {
      if (options == null) {
        options = {};
      }
      console.log("Start BLE")
      **bleManager.start(options, (error) =>** {
        if (error) {
          reject(error);
        } else {
          fulfill();
        }
      });
    });
  }
```

I am not sure what went wrong or how to fix it.

marcosinigaglia, 2017-08-04:
Hi, try to install manually.

Ideasrefined, 2017-08-04:
Update: Compiling and running included example results in the same error.

now moving to installing it manually But I am skeptical.

Ideasrefined, 2017-08-04:
Update #2:
Tried installing / linking using manual procedure described in the readme. After all the steps when I run "react-native run-android" I get the following error :- 

```
/home/enigma/codes/react/native/innovateBLE/node_modules/react-native-ble-manager/android/src/main/java/it/innove/BleManagerPackage.java:26: error: method does not override or implement a method from a supertype
        @Override
        ^
Note: /home/enigma/codes/react/native/innovateBLE/node_modules/react-native-ble-manager/android/src/main/java/it/innove/LegacyScanManager.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
1 error
:react-native-ble-manager:compileReleaseJavaWithJavac FAILED

FAILURE: Build failed with an exception.
```

marcosinigaglia, 2017-08-04:
Try with RN 0.46, I don't know if there are breaking changes in 0.47

virdesai, 2017-08-11:
Change `public class BleManagerPackage implements ReactPackage {

	public BleManagerPackage() {}

	@Override
	public List<NativeModule> createNativeModules(ReactApplicationContext reactApplicationContext) {
		List<NativeModule> modules = new ArrayList<>();

		modules.add(new BleManager(reactApplicationContext));
		return  modules;
	}

	**@Override
	public List<Class<? extends JavaScriptModule>> createJSModules() {
		return new ArrayList<>();
	}**

	@Override
	public List<ViewManager> createViewManagers(ReactApplicationContext reactApplicationContext) {
		return Collections.emptyList();
	}
}`

to

`public class BleManagerPackage implements ReactPackage {

	public BleManagerPackage() {}

	@Override
	public List<NativeModule> createNativeModules(ReactApplicationContext reactApplicationContext) {
		List<NativeModule> modules = new ArrayList<>();

		modules.add(new BleManager(reactApplicationContext));
		return  modules;
	}

	**public List<Class<? extends JavaScriptModule>> createJSModules() {
		return new ArrayList<>();
	}**

	@Override
	public List<ViewManager> createViewManagers(ReactApplicationContext reactApplicationContext) {
		return Collections.emptyList();
	}
}`

Ideasrefined, 2017-08-15:
Works as is for RN 0.46

closed 2017-08-15 by Ideasrefined

rgstephens, 2017-08-18:
I've also run into this issue with 0.47. Seems like this issue should be left open.

nomoreboredom, 2018-05-08:
For future reference, RN version 0.53, Typescript, I needed to import as ```import * as BleManager from 'react-native-ble-manager';```

dcmayankavi, 2020-04-20:
I'm facing same problem with below configuration:

"react": "16.11.0",
"react-native": "0.62.2",
"react-native-ble-manager": "7.2.0"

Android version: 10

shouer, 2020-06-17:
I'm facing same problem with below configuration:

"react": "16.8.3",
"react-native": "0.59.5",
"react-native-ble-manager": "^7.3.0",

Android version: 10

asukiaaa, 2024-06-16:
It depended on importing way in my case.

```ts
// import * as BleManager from "react-native-ble-manager" // Bad
import BleManager from "react-native-ble-manager" // OK

BleManager.start()
```

```json
    "react": "18.2.0",
    "react-native": "0.74.2",
    "react-native-ble-manager": "^11.5.4"
```