innoveit/react-native-ble-manager #193: android 4.4 fail to connect with error status 133

AutumnDog, 2017-08-04:
### Version

Tell us which versions you are using:

- react-native-ble-manager v5.1.1
- react-native v0.46
- iOS/Android v.4.4.2

sometimes when i scan and connect to a ble device, the connect function returns nothing, the promise dont reject or fulfill.
debug on the native android code ,the peripheral.onConnectionStateChange() function shows 
"onConnectionStateChange from 133 to 2 on peripheral"

i tried to take gatt.close(); and reconnect to the peripheral, does not help

marcosinigaglia, 2017-08-19:
Hi, the 133 code is an error, probably is not handled correctly in the connection.

AutumnDog, 2017-08-27:
@marcosinigaglia Thanks for reply! 
I found the error code 0x85 for GATT_ERROR in the android code, and catched the error. However it seems I can never connect to the device again forever , when I can connect to another device.
So what can I do to continue after this error ?

marcosinigaglia, 2017-08-28:
Do you investigate in the error type? If you restart the bluetooth (switch off and on) in the phone after the error you can connect?

thisboyiscrazy, 2018-03-06:
I am also getting 
```
onConnectionStateChange to 0 on peripheral: (mac) with status133
```
When trying to connect.

It's error code: 133

A lot, I do not know if it is related. But I do not have a solution yet.

thisboyiscrazy, 2018-03-06:
Found this I don't know if it is related (I am using testing on a samgsung)
```

        /*
         * In some implementations (i.e. Samsung Android 4.3) calling BluetoothDevice.connectGatt()
         * from thread other than main thread ends in connecting with status 133. It's safer to make bluetooth calls
         * on the main thread.
         */
```

closed 2018-06-14 by marcosinigaglia