innoveit/react-native-ble-manager #194: Descriptor values always null

hkotkanen, 2017-08-04:
### Version

Tell us which versions you are using:

- react-native-ble-manager commit 7ad211e40724fa45116c478db1b2b8cc67328cd1
- react-native v0.44.0
- Android v.7.0, Moto G4 Plus

### Expected behaviour

Values of characteristic descriptors should not be null. I'm running a custom peripheral, and have defined descriptors with UUID 2901 with a string value for all characteristics. I have verified with a 3rd party app ("BLE Scanner") that my peripheral is in fact sending a string as the value of a descriptor with UUID 2901. Screenshot from the 3rd party app:
[image]
I'm wondering if I've done something wrong, because in this 3rd party app interface the descriptor value must be explicitly retrieved by a read operation. Quickly looking over the native android code though, it seems like the descriptors array is populated by read operations on creation and that I don't need to explicitly make calls to read them separately in the JS code. I assume the relevant lines start [here](https://github.com/innoveit/react-native-ble-manager/blob/master/android/src/main/java/it/innove/Peripheral.java#L156), but might not understand what they actually do.

### Actual behaviour

Having `console.log(characteristic.descriptors);` in my app outputs into the log (output is the same for all characteristics):
`[ { value: null, uuid: '2902' }, { value: null, uuid: '2901' } ]`

### Steps to reproduce

1. Connect to device
2. Retrieve services
3. Examine descriptors

### Stack trace and console log

No error, values are simply null

marcosinigaglia, 2017-08-06:
Hi, with the actual code you need to read the characteristic, I'll check if you already have the value during the characteristics enumeration. I think the app read automatically the values.

hkotkanen, 2017-08-07:
@marcosinigaglia thanks for responding! I'm having a bit of a hard time understanding this so I appreciate your help a lot.

For clarity, here is a bit more of how I'm using `react-native-ble-manager`. I have in the my app
```
BleManager.connect(device.id).then(()=>{
    BleManager.retrieveServices(device.id).then((data)=>{
        console.log(data);
    })
})
```
which outputs the lines below. Is there something else that I can do to get the descriptor values? A document 
```
{ characteristics: 
 [ { properties: { Read: 'Read' },
     characteristic: '2a00',
     service: '1800' },
   { properties: { Read: 'Read' },
     characteristic: '2a01',
     service: '1800' },
   { descriptors: [ { value: null, uuid: '2902' } ],
     properties: { Indicate: 'Indicate' },
     characteristic: '2a05',
     service: '1801' },
   { descriptors: [ { value: null, uuid: '2902' }, { value: null, uuid: '2901' } ],
     properties: { Notify: 'Notify', Read: 'Read' },
     characteristic: '2a6e',
     service: '181a' },
   { descriptors: [ { value: null, uuid: '2902' }, { value: null, uuid: '2901' } ],
     properties: { Notify: 'Notify', Read: 'Read' },
     characteristic: '2a6f',
     service: '181a' },
   { descriptors: [ { value: null, uuid: '2902' }, { value: null, uuid: '2901' } ],
     properties: { Notify: 'Notify', Read: 'Read' },
     characteristic: '2a6d',
     service: '181a' },
   { descriptors: [ { value: null, uuid: '2902' }, { value: null, uuid: '2901' } ],
     properties: { Notify: 'Notify', Read: 'Read' },
     characteristic: '00000025-947d-41ef-a73e-73acf7a0de14',
     service: '0d52fc21-90ae-400f-b630-8e17630367c2' },
   { descriptors: [ { value: null, uuid: '2902' }, { value: null, uuid: '2901' } ],
     properties: { Notify: 'Notify', Read: 'Read' },
     characteristic: '00000010-947d-41ef-a73e-73acf7a0de14',
     service: '0d52fc21-90ae-400f-b630-8e17630367c2' } ],
services: 
 [ { uuid: '1800' },
   { uuid: '1801' },
   { uuid: '181a' },
   { uuid: '0d52fc21-90ae-400f-b630-8e17630367c2' } ],
rssi: -61,
advertising: 
 { bytes: 
    [ 2,
      ...
      0 ],
   data: 'Ag...AA=',
   CDVType: 'ArrayBuffer' },
id: 'my-peripheral-id',
name: 'my-peripheral-name }
```

marcosinigaglia, 2017-08-19:
Hi, i think you have to read manually the two characteristics 00000010-947d-41ef-a73e-73acf7a0de14 and 00000025-947d-41ef-a73e-73acf7a0de14 to retrieve the value you see in the other app.

hkotkanen, 2017-08-19:
Thank you but HOW would I do that? A JS method I have to call on a connected peripheral? Or do I need to implement it myself in the Java side of this lib? I can look at it and submit a PR if that's the case, but it would be great if you could at least give me a direction where to start looking.

marcosinigaglia, 2017-08-21:
You have to connect, call retrieveServices and then call read on one of the characteristic and then in the other.

closed 2017-08-29 by marcosinigaglia