innoveit/react-native-ble-manager #200: id & mac question

Symous, 2017-08-22:
Hi,awesome lib!

I'm using getDiscoveredPeripherals to get all discovered ble.

But there is a question, I use android and iphone to scan a same ble, but they get different results, it returns "mac" on android but "id" on ios.

Is there any way I can get a same "result string" to represent the ble ?Thanks a lot!

henninghall, 2017-08-22:
You can't get the mac address from an ios device (or in android M+). Also, ID's read by ios devices varies between different ios devices.  

I don't know the best approach to solve this but one way is putting an ID in the advertisement data.

Symous, 2017-08-23:
@henninghall In fact, what I'm doing with mac address on iOS is just same as what you said.Just looking for any better way to achieve this. Thank you!

closed 2017-08-23 by Symous