innoveit/react-native-ble-manager #202: Can I take 'XXXX' as the parameter of serviceUUID and characteristicUUID?

zhanguangao, 2017-08-23:
start the notification：
`BleManager.startNotification(peripheralId, serviceUUID, characteristicUUID)
`

[image]

Can I take '1801' as  the parameter  of serviceUUID , take '2A23' as  the parameter  of characteristicUUID, because the returned UUID parameters  are four strings.
I try it，but fail.

marcosinigaglia, 2017-08-29:
Hi, are you sure that the notification is supported on this characteristic?

zhanguangao, 2017-08-29:
@marcosinigaglia  All of the serviceUUID and characteristicUUID I got are the type of 'XXXX',I can't start notification and write.does it support this bluetooth?

marcosinigaglia, 2017-08-29:
In your example the 2A50 characteristic is only writable, you have to check the 2A23 characteristic.

zhanguangao, 2017-08-29:
but get error when I execute the BleManager.write().

KarthickCSK, 2017-09-27:
Im using BLE simulator and im getting serviceUUID and characteristicsUUID of type XXXX. When i try to call a read funct with characteristics belongs to readable it throws error. ie, Chracteristics not found.

abikalwa, 2017-10-25:
I too face the same error. Unable to Read the characteristic Data.

zhanguangao, 2017-10-30:
I See.They are the base UUID '0x0000xxxx-0000-1000-8000-00805F9B34FB',So we can replace 'xxxx' manually.If we get '1801', the UUID is '0x00001801-0000-1000-8000-00805F9B34FB'.

closed 2018-01-11 by marcosinigaglia