innoveit/react-native-ble-manager #203: handleDiscoverPeripheral Called only once after that never get called??

AshokICreate, 2017-08-24:
```
 import React, { Component } from 'react';
import {
  AppRegistry,
  StyleSheet,
  Text,
  View,
  TouchableHighlight,
  NativeAppEventEmitter,
  NativeEventEmitter,
  NativeModules,
  Platform,
  PermissionsAndroid,
  ListView,
  ScrollView
} from 'react-native';
import Dimensions from 'Dimensions';
import BleManager from 'react-native-ble-manager';
import TimerMixin from 'react-timer-mixin';
import reactMixin from 'react-mixin';

const window = Dimensions.get('window');
const ds = new ListView.DataSource({rowHasChanged: (r1, r2) => r1 !== r2});

const BleManagerModule = NativeModules.BleManager;
const bleManagerEmitter = new NativeEventEmitter(BleManagerModule);

export default class App extends Component {
  constructor(){
    super()

    this.state = {
      scanning: false,
      peripheral: null,
    }

    this.handleDiscoverPeripheral = this.handleDiscoverPeripheral.bind(this);
    this.handleStopScan = this.handleStopScan.bind(this);
    this.handleUpdateValueForCharacteristic = this.handleUpdateValueForCharacteristic.bind(this);
    this.handleDisconnectedPeripheral = this.handleDisconnectedPeripheral.bind(this);
  }

  componentDidMount() {
    BleManager.start({showAlert: false, allowDuplicates: true});

    this.handlerDiscover = bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral );
    this.handlerStop = bleManagerEmitter.addListener('BleManagerStopScan', this.handleStopScan );
    this.handlerDisconnect = bleManagerEmitter.addListener('BleManagerDisconnectPeripheral', this.handleDisconnectedPeripheral );
    this.handlerUpdate = bleManagerEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', this.handleUpdateValueForCharacteristic );

    if (Platform.OS === 'android' && Platform.Version >= 23) {
        PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION).then((result) => {
            if (result) {
              console.log("Permission is OK");
            } else {
              PermissionsAndroid.requestPermission(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION).then((result) => {
                if (result) {
                  console.log("User accept");
                } else {
                  console.log("User refuse");
                }
              });
            }
      });
    }
    this.startScan();
  }

  componentWillUnmount() {
    if(this.state.peripheral!=null) {
      BleManager.disconnect(this.state.peripheral.id);
    }
    this.handlerDiscover.remove();
    this.handlerStop.remove();
    this.handlerDisconnect.remove();
    this.handlerUpdate.remove();
  }

  startScan() {
    if (!this.state.scanning) {
      BleManager.scan([], 3, false).then((results) => {
        this.setState({scanning:true});
      });
    }
  }

  handleDiscoverPeripheral(peripheral){
    if (peripheral.name) {
      if (
        peripheral.name.toLowerCase().indexOf('T') === 0 &&
        peripheral.advertising.kCBAdvDataServiceUUIDs &&
        peripheral.advertising.kCBAdvDataServiceUUIDs.length
      ) {
        console.log(`Peripheral Found and Data is:-- ${JSON.stringify(data)}`);
        this.setState({peripheral: peripheral})
        BleManager.connect(peripheral.id).then(() => {
          this.setTimeout(() => {
            BleManager.retrieveServices(peripheral.id).then((peripheralInfo) => {
              this.setTimeout(() => {
                BleManager.startNotification(peripheral.id, '6E400001-B5A3-F393-E0A9-E50E24DCCA9E', '6E400004-B5A3-F393-E0A9-E50E24DCCA9E').then(() => {
                  console.log('Started notification on ' + peripheral.id);

                }).catch((error) => {
                  console.log('Notification error', error);
                });
              }, 1000);
              this.setTimeout(() => {
                BleManager.startNotification(peripheral.id, '6E400001-B5A3-F393-E0A9-E50E24DCCA9E', '6E400005-B5A3-F393-E0A9-E50E24DCCA9E').then(() => {
                  console.log('Started notification on ' + peripheral.id);

                }).catch((error) => {
                  console.log('Notification error', error);
                });
              }, 1100);
              this.setTimeout(() => {
                BleManager.startNotification(peripheral.id, '6E400001-B5A3-F393-E0A9-E50E24DCCA9E', '6E400006-B5A3-F393-E0A9-E50E24DCCA9E').then(() => {
                  console.log('Started notification on ' + peripheral.id);

                }).catch((error) => {
                  console.log('Notification error', error);
                });
              }, 1200);
            });

          }, 900);
        }).catch((error) => {
          console.log('Connection error', error);
        });
      }
    }
  }

  handleUpdateValueForCharacteristic(data) {
    console.log('Received data from ' + data.peripheral + ' characteristic ' + data.characteristic, data.value);
  }

  handleStopScan() {
    console.log('Scan is stopped');
    this.setState({ scanning: false });
  }

  handleDisconnectedPeripheral(data) {
    removePeripheral(data.peripheral.peripheralId)
    console.log('Disconnected from ' + data.peripheral);
  }

  render() {
    return (
      null
    );
  }
}

reactMixin(App.prototype, TimerMixin);

```

AshokICreate, 2017-08-28:
```
RCT_EXPORT_METHOD(start:(NSDictionary *)options callback:(nonnull RCTResponseSenderBlock)callback)
{
    NSLog(@"BleManager initialized");
    NSMutableDictionary *initOptions = [[NSMutableDictionary alloc] init];

    if ([[options allKeys] containsObject:@"showAlert"]){
        [initOptions setObject:[NSNumber numberWithBool:[[options valueForKey:@"showAlert"] boolValue]]
                        forKey:CBCentralManagerOptionShowPowerAlertKey];
    }

    if ([[options allKeys] containsObject:@"restoreIdentifierKey"]) {
        [initOptions setObject:[options valueForKey:@"restoreIdentifierKey"]
                        forKey:CBCentralManagerOptionRestoreIdentifierKey];

    if (_sharedManager) {
        manager = _sharedManager;
        manager.delegate = self;
    } else {
        manager = [[CBCentralManager alloc] initWithDelegate:self
                                                       queue:dispatch_get_main_queue()
                                                     options:initOptions];
        _sharedManager = manager;
      }
    } else {
        manager = [[CBCentralManager alloc] initWithDelegate:self
                                                       queue:dispatch_get_main_queue()
                                                     options:initOptions];
        manager.delegate = self
    }

    callback(@[]);
}
```

Modified To

```
RCT_EXPORT_METHOD(start:(NSDictionary *)options callback:(nonnull RCTResponseSenderBlock)callback)
{
    NSLog(@"BleManager initialized");
    NSMutableDictionary *initOptions = [[NSMutableDictionary alloc] init];

    if ([[options allKeys] containsObject:@"showAlert"]){
        [initOptions setObject:[NSNumber numberWithBool:[[options valueForKey:@"showAlert"] boolValue]]
                        forKey:CBCentralManagerOptionShowPowerAlertKey];
    }

    if ([[options allKeys] containsObject:@"restoreIdentifierKey"]) {
        [initOptions setObject:[options valueForKey:@"restoreIdentifierKey"]
                        forKey:CBCentralManagerOptionRestoreIdentifierKey];

    if (_sharedManager) {
        manager = _sharedManager;
        manager.delegate = self;
    } else {
        manager = [[CBCentralManager alloc] initWithDelegate:self
                                                       queue:dispatch_get_main_queue()
                                                     options:initOptions];
        _sharedManager = manager;
        manager.delegate = self; ////Added
      }
    } else {
        manager = [[CBCentralManager alloc] initWithDelegate:self
                                                       queue:dispatch_get_main_queue()
                                                     options:initOptions];
        manager.delegate = self //Added
    }

    callback(@[]);
}
```

Than it works.

marcosinigaglia, 2017-08-28:
The CBCentralManager allocation use the initWithDelegate parameter so your changes are not useful.

AshokICreate, 2017-08-28:
Thanks for the input.

`else {
        manager = [[CBCentralManager alloc] initWithDelegate:self
                                                       queue:dispatch_get_main_queue()
                                                     options:initOptions];
       _sharedManager = manager; //This line has to be added in else condition right?
    }`

`**_sharedManager = manager; //This line has to be added in else condition right otheriwise _sharedManager is null**`

marcosinigaglia, 2017-08-29:
It's already fixed.

closed 2017-08-29 by marcosinigaglia