innoveit/react-native-ble-manager #204: Can you tell me if encryption handled in the software or does the hardware have this feature built in?

anwarhamr, 2017-08-24:
My solution requires that the data to be encrypted.  I will create a custom GATT service similar to the User Data service (https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.user_data.xml)
How is encryption handled with this library and what are the steps for receiving that data?

Cheers,
Rob

marcosinigaglia, 2017-08-28:
Hi, I think is handled from the OS so I think is transparent for this library. Try and let me know.

anwarhamr, 2017-08-28:
But how would I specify that it should be encrypted?  Not all communication
between the devices is encrypted so how would one specify that it should?

Cheers,
Rob Lingstuyl
Software (Programmer & Architect)
+1 801 599-1596
808numbers@gmail.com
http://www.linkedin.com/in/robertl

*Somebody should tell us, right at the start of our lives, that we are
dying. Then we might live life to the limit, every minute of every day. Do
it! I say. Whatever you want to do, do it now! There are only so many
tomorrows.* ~ Pope Paul VI

marcosinigaglia, 2017-08-29:
I think is the BLE device that can request the secure connection.

closed 2017-10-09 by marcosinigaglia