innoveit/react-native-ble-manager #205: Different name on iOS / Android

mateusz-, 2017-08-25:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.0.0
- react-native v0.47.1
- iOS/Android v7.1.1 (Android) 10.3.3 (iOS)

### Expected behaviour

When scanning for devices (```BleManagerDiscoverPeripheral``` event) the name should be the same on both iOS and Android.

### Actual behaviour

The name may be different.

### Steps to reproduce

Set the shortened local name (or complete name) to one value. Set Generic Access service device name to another value. Android will always use the former. iOS will always use the latter.

The BLE spec states that the two should be the same, if using complete name (CSS v7 sec. 1.1.2) - so it should not be a problem under ideal circumstances. However, the maximum advertised payload is 31 bytes, which means that shortened local name can be used instead. This can cause the discrepancy.

Is it possible to use the Generic Access service device name, on both platforms, to avoid this issue?

### Stack trace and console log

iOS
```json
{
  "advertising": {
    "kCBAdvDataIsConnectable": true,
    "kCBAdvDataLocalName": "Dock Con",
    "kCBAdvDataServiceUUIDs": ["02FF5604-BA5E-F4EE-5CA1-EB1E5E4B1CE0"]
  },
  "id": "5CD075B0-A875-4B3B-9B98-C3AC75E4D76F",
  "name": "Dock Controller",
  "rssi": -71
}
```

Android
```json
{
  "advertising": {
    "CDVType": "ArrayBuffer",
    "bytes": ["..."],
    "data": "..."
  },
  "id": "E3:4F:0C:93:54:94",
  "name": "Dock Con",
  "rssi": -61
}
```

marcosinigaglia, 2017-08-28:
Hi @mateusz- , I look into your problem and what you need is a characteristic value so you have to connect and retrieve manually, is not possible to have this value during the scanning.

mateusz-, 2017-08-28:
Thanks for getting back to me @marcosinigaglia.

I did some more reading - besides advertising data, there is another packet called scan response data. This may be used when all of the data does not fit within the advertising packet.

As a side note - the "data" property in the exposed advertising object has a length of 62 bytes - the first 31 bytes is the advertising data, while the last part is the scan data. I didn't have to parse it - the library handles it properly and the name is set automatically.

So, this basically solves my problem, however, the behavior is still odd, with iOS getting its data from the Generic Access service.

I'll close this, thanks.

closed 2017-08-28 by mateusz-

marcosinigaglia, 2017-08-29:
Yes, it's strange but for what I tried the BLE is one of the worst implemented standard :)