innoveit/react-native-ble-manager #206: BleManager.removePeripheral(peripheral.id) is not a function error?

AshokICreate, 2017-08-28:
"react-native-ble-manager": "6.0.0",

BleManager.removePeripheral(peripheral.id) is not a function error?

stephenjen, 2017-08-29:
@AshokICreate I just tried this and it seems the error is only in iOS for me.  Looking at the merge I'm not sure if it was implemented for iOS.

marcosinigaglia, 2017-08-29:
In iOS is not implemented.

stephenjen, 2017-08-29:
I'm having the issue where reconnect does not work in iOS when the my code follows these steps:

1 Scan
2 Connect
3 Get and save peripheral ID
4 StopScan
5 Disconnect
6 Connect
7 WriteWithoutReponse

In Android, the when I turn off the BLE device and then turn it back on, it reconnects.  In iOS, however, it does not reconnect.  iOS if I restart the app, which starts fro step 6 above, it can connect and reconnect without problem.  

What might be the problem?

marcosinigaglia, 2017-08-29:
Do you have some error?

stephenjen, 2017-08-29:
The steps I described generates no error, just cannot reconnect in iOS if scanning occurred first.  I thought BleManager.removePeripheral might do the trick, but as you said it's not implemented in iOS.

marcosinigaglia, 2017-08-29:
Did you saw something in the device log?

stephenjen, 2017-08-29:
After the disconnect listener is handled there's nothing else.

kf729ik, 2017-11-24:
Same issue like @stephenjen, IOS cannot reconnect, if I restart the app it works fine.

stephenjen, 2017-11-24:
@kf729ik my issue was due to calling start() more than one times.  I had to restructure all my BLE operations to within one action file so I wouldn't need to call start() more than once.

kf729ik, 2017-11-24:
@stephenjen I think that I'm facing the same issue, so I'm going to try out this approach, thanks for the comment.

closed 2017-11-27 by marcosinigaglia