innoveit/react-native-ble-manager #208: Subscribing to multiple characteristics, an example please.

anwarhamr, 2017-08-30:
Hello I see this [issue 79](https://github.com/innoveit/react-native-ble-manager/issues/79) is closed but the code posted is really ugly.  Additionally, it does not make sense why there are multiple calls to addListener since none of the callbacks are specific to the characteristic value.

I have a device that publishes a single service w/ 3 characteristics.  I can subscribe to each individually but get this message when I try to do multiple.  
console output:
Starting notification for char 9c16c3ca-1f05-4089-a0af-aa8c921183ca
Starting notification for char c4ddc87d-a470-4606-b20a-6eff9444c725
Starting notification for char 5eb2fd1d-3310-4b67-b09e-933891467fdd
Notification error Failed to set client characteristic notification for c4ddc87d-a470-4606-b20a-6eff9444c725 
Notification error Failed to set client characteristic notification for 5eb2fd1d-3310-4b67-b09e-933891467fdd

The first characteristic I subscribe to always works but not the subsequent.

I've added setTimeout but that doesn't seem to work either.  Can we get an example?  I'll even update the docs from this issue if someone can help.

marcosinigaglia, 2017-09-08:
Hi, I'm sorry but the calls are asyncronous and you can't call multiple function at the same time, is how the module work and you need to handle this in your project.

todorone, 2017-12-20:
@marcosinigaglia I've also stuck with subscribing to notifications of 3 characteristics. While single one works perfect, I get errors when trying to subscribe to all of them.
So the question is - is it possible to subscribe to 3 characteristics at the same time using `BleManager`?

```
    // Notifications
    await BleManager.startNotification(id, service, characteristicCartridge)
    await BleManager.startNotification(id, service, characteristicTestResult)
    await BleManager.startNotification(id, service, characteristicTestError)
```
Unfortunately this code doesn't works for me, although I'm not calling all functions at the same time, but invoke them consequentially.
Thanks in advance.

marcosinigaglia, 2017-12-20:
You can't subscribe in the same time, in my experience is also better to add delay beetween these operations.

closed 2018-01-11 by marcosinigaglia

VitamineC, 2018-03-13:
There is my resolve:
```
function addNotificationListeners(device, notifications, current = 0) {
  const { UUID, errorCallback = () => {} } = notifications[current];
  return BleManager.startNotification(device, serviceUUID, UUID)
    .then(() => {
      if (current < notifications.length) {
        return addNotificationListeners(device, notifications, ++current);
      } else {
        return null;
      }
    })
    .catch(error => {
      errorCallback(error);
    });
}
```

anwarhamr, 2018-03-13:
@VitamineC have you tested this with multiple notification wireups at the same time?  I don't see a delay in your code unless it is elsewhere. 
I was able to get it working using a delay/setTimeout.  Here is my example but it's not so straight forward unless you understand Observables and promise's.

```
function registerNotifications(peripheralId, service, notificationChars){
    this.log.info('BLE requesting notification for ', peripheralId, notificationChars);
    return Observable.from(notificationChars)
      .map((characteristic)=> {
        this.log.info('BLE notification starting ', peripheralId,characteristic);

        const s = Observable
          .defer(()=>BleManager.startNotification(peripheralId,service,characteristic))
          .delay(500)
          .do(val=>this.log.info('notification set',characteristic))
          .mapTo(Observable.of({id:peripheralId,service, characteristic,success:true}))
          .retryWhen(errors=>
            errors
              .do(val=>{
                if (this.shouldRetryError(val)) throw val;
                this.log.info('retry notification error',characteristic,val);
              })
              .delay(300))
          .catch(()=>({id:peripheralId,service, characteristic, success:false}));
        return   s.first();
      });
  }
```

VitamineC, 2018-03-13:
@anwarhamr  I tested it with multiple notification at the same time. I use recursion, I try to setup new notification only after previous sent success (then) to me.  At each step, I take the next notification and so on until I pass through the all transferred array.
In fact, I copy your map, but only instead of delay, I'm waiting for the successful installation of the past.

About your example:
I think that using delay - bad practise.
What's happen with your programm if something will happened and delay be more than 500? You will not able to startNotification.

ethanyuwang, 2020-01-15:
Has anyone figured out a good way to do this?

anwarhamr, 2020-01-16:
@ethanyuwang  ya the nesting thing works best.

mim-Armand, 2020-07-08:
fiy, looks like this problem exists only on Android, in IOS I could subscribe to as many notifications as I need, all at the same time with no problem.