innoveit/react-native-ble-manager #210: Notification doesn't trigger BleManaterDidUpdateValueForCharacteristic Event

w8choi, 2017-09-01:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.1.0
- react-native v0.47.2
- iOS/Android

### Expected behaviour

When starting notification, receive data sent by BLE device ( Event triggered)

### Actual behaviour

Starts notification but nothing comes in from BLE device. ( Event is not triggered )
When I use it with the app that's using older version of RNBleManager, I can see the data coming in.

### Steps to reproduce

1. connect to device
2. start notification
3. see if it receives any data

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2017-09-08:
Hi, the problem start with RN 0.47 or with the new library version?

w8choi, 2017-09-08:
It had that issue with library ver. 5.1.1 on RN 0.45.1 with iOS 10.X and since I updated both RN and library, it stopped getting any notification after starting notification.

marcosinigaglia, 2017-09-08:
Do you use the NativeEventEmitter?

w8choi, 2017-09-08:
Yes I do.
So like my set up is pretty similar to your set up in your example except I'm not writing anything right after I start notification

marcosinigaglia, 2017-09-14:
I tested with RN 0.48.0 without problem.

closed 2017-09-14 by marcosinigaglia