innoveit/react-native-ble-manager #211: 'React/RCTBridgeModule.h' file not found

aeqgg, 2017-09-03:
After add react-native-ble-manger to an ejected CRNA project, builds fail.

### Version

Tell us which versions you are using:

    "expo": "^20.0.0",
    "react": "16.0.0-alpha.12",
    "react-native": "https://github.com/expo/react-native/archive/sdk-20.0.0.tar.gz",
    "react-native-ble-manager": "^6.1.0"
    iOS simulator

### Expected behaviour

Since this is a new project with all tools newly installed, it should work as per the instructions and compile properly.

### Actual behaviour

The compile fails as described above.

### Steps to reproduce

Start from a brand new create-react-native-app application, then ejected. Ran pod install and ensured that the resulting application still compiled in Xcode and ran properly.

Then, ran:

    npm i --save react-native-ble-manager
    react-native link react-native-ble-manager

and tried to compile again. I get the message

    'React/RCTBridgeModule.h' file not found

This occurs in the BleManager.m file.

### Stack trace and console log

marcosinigaglia, 2017-09-08:
Hi, I think is an installation problem.

marcosinigaglia, 2017-09-14:
I see now that you use expo, you can't use external module.

closed 2017-09-14 by marcosinigaglia

jamesguan, 2017-11-30:
I'm trying to build in xcode without expo and I'm having the same issue with the example code. Any reasons why this would come up? I have am using the latest version of react native as well.

Glendonh, 2018-01-30:
It looks like the original poster is detached from Expo, which means that external modules can be used. I'm also detached from Expo and I'm having the same issue. Do you have any insight into this? Also, @aeqgg did you ever work past this?