innoveit/react-native-ble-manager #212: Android Pairing Issues

mateusz-, 2017-09-05:
### Version

- react-native-ble-manager v6.0.0
- react-native v0.47.1
- iOS/Android 6.0.1 (Android) 10.3.3 (iOS)

### Expected behaviour

[Writing to a characteristic](https://github.com/innoveit/react-native-ble-manager#writeperipheralid-serviceuuid-characteristicuuid-data-maxbytesize) requiring authentication should trigger a pairing request and a promise rejection on authentication failure.

On authentication failure, subsequent writes should error with a valid message.

### Actual behaviour

On iOS everything works well:
A pairing prompt is presented upon trying to write to a characteristic requiring authentication (good?). If an invalid passkey is provided, or if cancel is pressed, write then rejects with an ```Encryption is insufficient.``` error (good?). Subsequent writes fail with ```Authentication is insufficient.``` error (good?).

If a valid passkey is entered, subsequent writes resolve (good?).

On Android:
A pairing prompt is presented upon trying to write to a characteristic requiring authentication (good?). If an invalid passkey is provided, or if cancel is pressed, write does not reject (bad?). Subsequent writes fail with ```You're already writing``` error (bad?). 

If a valid passkey is entered, subsequent writes reject with ```You're already writing``` (bad?).

I added some breakpoints to the Android source code. ```gatt.writeCharacteristic(characteristic)``` succeeds but ```onCharacteristicWrite``` never gets called. ```writeCallback``` does not get set to null.

marcosinigaglia, 2017-09-08:
Hi, I can't simulate so I can't understand the problem, a PR will be very helpful. I think the iOS behaviors are ok so we need to replicate it on Android.

whitedogg13, 2017-09-09:
I'm interested in this issue as well. I also suffer a lot when handling secure BLE + Android platform.

@marcosinigaglia I think a reliable way to dealing with this is to bridge the `createBond` API to JavaScript, let the app itself control whether to use any Bluetooth secure feature.
* https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#createBond()

With this API, app can make sure the secure BLE link is setup then proceed following characteristics read/write. 

I think this is quite important and I will try to implement this when I'm available. 

BTW, @mateusz- I'm also wondering why the Android didn't fire any error when accessing a characteristic requiring auth. It seems that Android simply ignore the write command without complaining. Is this issue happens on certain device or all devices?

mateusz-, 2017-09-09:
I only tried it on a single device (some generic tablet with Android 6.0.1).

I was able to get around this issue by triggering the initial pairing prompt by using read() instead of write() (writeWithoutResponse() also works). I believe this is an Android bug (that is, onCharacteristicWrite callback not being invoked), but I can't really find anything about it.

I think explicit bonding would be nice, if supported on both iOS and Android, but iOS does not allow that, right? Personally, I like to keep workflows the same across platforms but I can definitely see a counter-argument to that.

whitedogg13, 2017-09-10:
@mateusz- Yes, you are right about iOS, which provides no explicit bonding API and will always trigger pair/bond when first accessing secured GATT characteristics.

Philipwzj, 2017-09-13:
@whitedogg13 How's the progress? Can I help with anything? : )

whitedogg13, 2017-09-14:
@Philipwzj Currently I have a WIP version here: https://github.com/whitedogg13/react-native-ble-manager/tree/android-create-bond. However, I don't have enough time to verify it yet, so It will be great if someone can help with that!

whitedogg13, 2017-09-21:
I finally got one bondable device to test, and the [PR](https://github.com/innoveit/react-native-ble-manager/pull/216) is just submitted.

closed 2017-09-25 by marcosinigaglia

Philipwzj, 2017-09-25:
@whitedogg13 Thank you so much! 
When I'm testing the createBond function with two BLE devices at the same time, RN gave me an error of "Illegal callback invocation from native module. The callback type only permits a single invocation from native code". Any suggestions on what I should do? Thank you!

whitedogg13, 2017-09-26:
Hi @Philipwzj , actually we do have protection against bonding two devices at the same time, in such a case you should receive a promise rejections says "Only allow one bond request at a time".

I will suggest you check the test code, to see whether the error comes from createBond or some other calls. (the error basically means each callback passed from JS to native can be invoked at least once)

However, in this project we generally don't deal with native thread synchronization, so maybe it's still chance to have some race condition. If it's the case, you can consider adding java `synchronized` guard.

Thanks!

mateusz-, 2017-10-24:
I just tried using the ```createBond``` function. What I observed was:

1. Peripheral sends a slave security request once central connects.
2. Central starts the pairing process automatically as a response to a slave security request.
3. Central starts bonding with ```createBond```.
4. Central ```createBond``` promise rejects with ```Create bond request fail```.

I think if I got rid of the slave security request everything would work (it is optional), but pairing would not start automatically on iOS - it would have to be triggered by reading or writing to a characteristic.

whitedogg13, 2017-10-26:
@mateusz- Hi, what platform are you using? The `createBond` API is an Android only one. 

Currently there seems no way to do explicit bond for iOS. So what you describe is correct and exactly the way iOS work for bonding:
> it would have to be triggered by reading or writing to a characteristic.

Or more explicitly:
> it would have to be triggered by reading or writing to a **protected** characteristic.

Apple says in https://developer.apple.com/hardwaredrivers/BluetoothDesignGuidelines.pdf:
> The accessory should not request pairing until an ATT request is rejected using the Insufficient Authentication error code ... If, for security reasons, the accessory requires a bonded relationship with the Central, the Peripheral should reject the ATT request using the Insufficient Authentication error code, as appropriate. As a result, the Apple product may proceed with the necessary security procedures.

So that result in the behavior you described!

Also, here is a extreme valuable post about BLE security. Highly recommended, just check it out!
* https://medium.com/@kbabcockuf/bridging-the-gap-bluetooth-le-security-653078852c24

mateusz-, 2017-10-26:
I understand all of that. The only thing that I am pointing out is that a central (iOS or Android, doesn't matter) can start a bonding process prior to reading or writing to a protected characteristic as a response to a slave security request (i.e., the peripheral notifies the central of security it requires once a central connects).

It doesn't look like Apple's guide mentions it, but here's an excerpt from Bluetooth specifications document (Version 4.2 [Vol 3, Part H, Section 2.4.6]):

> The slave device may request security by transmitting a Security Request command to the master. When a master device receives a Security Request command it may encrypt the link, initiate the pairing procedure, or reject the request.

What I observed was the bonding being initiated as a response to a slave security request, with ```createBond()``` being called shortly afterwards, which I believe is a problem. ```isBonding()``` or similar method is not exposed, so there's no way to tell if bonding is already taking place. Is this case being handled internally? Is it possible to handle it? I don't know - I haven't fully examined the source code, nor did I look through Android's Bluetooth API.

whitedogg13, 2017-10-26:
Thanks for the explaining, now I can understand your concern.

I think the scenario you described will indeed causes some issues, maybe we can try some alternatives: 
* Call `createBond` directly without prior `connect` call, since it will still connect to the peripheral first.
* Avoid calling `createBond` at all, and just leave it to the OS. Wait a while, and use `getBondedPeripherals` API (which I add to this library recently) to check if the peripheral you're targeting is bonded. 

BTW, I think the discussion we made here is valuable, dealing with secured BLE does require lots of extra considerations and insights. Thanks for that.

dreamlover, 2019-04-27:
I got the same error message "You're already writing" on android when I hit a button which writes data to a bluetooth characteristic repeatedly as fast as possible.
I found this code snippet in the write method of  Peripheral class:
`
if (writeCallback != null) {
	callback.invoke("You're already writing");
}
`
When I commented it out, my problem gone. Is it needed to wait for the success of the previous written to write again?

isThereAnyF-ingNameAvailable, 2020-05-15:
If you get this error, it happens because the write function locks write until the write "is complete".
If you use write() the BLE stack will wait for a response from the peripheral before opening up write again. One possible solution is to use async/try/await like this:
async ble_write_peripheral(mac_addr, svcUUID, writeUUID, value) {
  /* this is for strings. Convert your data to bytearray */
  binaryArray = Array.from(value, (char, i) => value.charCodeAt(i))
  try {
    await BleManager.write(mac_addr, svcUUID, writeUUID, binaryArray, binaryArray.length);
    console.log("BleManager.write: success");
  } catch (error) {
    console.log('BleManager.write: ', error);
  }
}

Another option is to use write without confirm, then the next write is open upon return of the function, i.e.
BleManager.writeWithoutResponse(mac_addr, svcUUID, writeUUID, binaryArray, binaryArray.length, null);

Sorry about the formatting, the code formatting and preview doesn't work right now.