innoveit/react-native-ble-manager #215: Not finding any device...

HighSoftWare96, 2017-09-20:
Hi,
I've tried this plugin, but it doesn't work on my phone (Huawei P9 Lite, Android 7).
I checked out all the possible problems but I'm not able to find out where the bug could be.
This is my app code:
```
import React, { Component } from 'react';
import { DrawerLayoutAndroid, AppRegistry, Text, View, Alert, StatusBar, TouchableHighlight, ToolbarAndroid, Image, ScrollView, Button, ActivityIndicator,
Platform, PermissionsAndroid, NativeAppEventEmitter } from 'react-native';
import { StackNavigator, } from 'react-navigation';
import { OneBtnMode } from './screens/oneBtnMode';
import { drawerStyles, mainStyles } from './styles/common.styles';
import { homeStyles } from './styles/home.styles';
import { CommonBars } from './common/common.components';
import { evaluateActionBarTitle } from './common/common.functions';
import { BluetoothService } from './services/bluetooth.service';

export class AppHome extends React.Component {

    constructor(props) {
        super(props);
        // ricevo l'emitter e gli aggiungo il listener per controllare lo stato on/off del ble == e altri eventi
        this.bleStateHandler = NativeAppEventEmitter.addListener('BleManagerDidUpdateState', this.handleEnablingBluetooth.bind(this));
        this.bleDiscoveredPer = NativeAppEventEmitter.addListener('BleManagerDiscoverPeripheral', this.handleNewDevice.bind(this));
        this.bleScanEnded = NativeAppEventEmitter.addListener('BleManagerStopScan', this.handleScanEnded.bind(this));

        // stato inizialmente nullo
        this.state = {
            blthOn: undefined,
            loading: false,
        };
        // controllo lo stato iniziale del bluetooth -- handleEnablingBluetooth gestirà l'evento
        BluetoothService.getStatus();
    }

    componentDidMount() {
        // inizializzo il modulo bluetooth una volta per tutte
        BluetoothService.initialize();
        // richiesta permessi localizzazione
        if (Platform.OS === 'android' && Platform.Version >= 23) {
            PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION).then((result) => {
                if (result) {
                    console.log("Permission is OK");
                } else {
                    PermissionsAndroid.requestPermission(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION).then((result) => {
                        if (result) {
                            Alert.alert("User accept");
                        } else {
                            Alert.alert("User refuse");
                        }
                    });
                }
            });
        }
    }

    static navigationOptions = {
        drawerLabel: 'Connect devices',
        drawerIcon: ({ tintColor }) => (
            <Image
                source={require('./images/homeIcon.png')}
                style={[drawerStyles.buttonIcon, { tintColor: tintColor }]}
            />
        ),
    };

    render() {
        const { navigate } = this.props.navigation;
        const { state } = this.props.navigation;

        return (
            <View style={{ flex: 1 }}>
                {/*Passo il navigatore alla barra principale in modo poi da poter aprire il drawer*/}
                <CommonBars navigate={navigate} title={evaluateActionBarTitle(state.routeName)} />
                <View style={homeStyles.mainPage}>
                    <View style={homeStyles.upContent}>
                        {/*Se lo stato indica che il bluetooth è spento allora mostrera l'immagine BN altrimenti quella blu*/}
                        <Image source={this.state.blthOn ? require('./images/blthOn.png') : require('./images/blthOff.png')} style={homeStyles.upContentImage} />
                        <View style={homeStyles.upContentButtons}>
                            {/*Stile e funzione determinati dallo stato del ble*/}
                            <TouchableHighlight underlayColor="#64B5F6" style={!this.state.blthOn ? homeStyles.upContentSingleBtn : homeStyles.upContentSingleBtnOff} onPress={this.state.blthOn ? null : this.enableBletooth}>
                                <View style={homeStyles.singleBtnContent}>
                                    <Image style={homeStyles.singleBtnIcon} source={require('./images/enable.png')} />
                                    <Text style={homeStyles.singleBtnText}>Enable bluetooth</Text>
                                </View>
                            </TouchableHighlight>
                            <TouchableHighlight underlayColor="#64B5F6" style={!this.state.loading && this.state.blthOn ? homeStyles.upContentSingleBtn : homeStyles.upContentSingleBtnOff} onPress={this.state.loading  && this.state.blthOn ? null : this.initScan.bind(this)}>
                                <View style={homeStyles.singleBtnContent}>
                                    {this.state.loading ? <ActivityIndicator color='white' /> : <Image style={homeStyles.singleBtnIcon} source={require('./images/scan.png')} />}
                                    <Text style={homeStyles.singleBtnText}>Scan</Text>
                                </View>
                            </TouchableHighlight>
                        </View>
                    </View>
                    <View style={homeStyles.downContent}>
                    </View>
                </View>
            </View>
        );

    }

    // FUNZIONI DEI BOTTONI
    // Funzione di abilitazione del BLE
    enableBletooth() {
        BluetoothService.enableBlth().then(() => {
        }).catch(() => {
            Alert.alert("Bluetooth", "Bluetooth off!");
        })
    };

    initScan() {
        BluetoothService.initScan().then(() => {
            console.log("Scanning....");
            this.setState(previousState => {
                // scan STARTED!
                return { loading: true };
            });
        })
    };

    // EVENT LISTENERS ====================================================
    // gestore eventi di cambiamento del bluetooth
    handleEnablingBluetooth(args) {
        if (args.state == 'on') {
            this.setState(previousState => {
                return { blthOn: true };
            });
        }
        else {
            this.setState(previousState => {
                return { blthOn: false };
            });
        }
    }

    // gestione evento di fine scansione
    handleScanEnded(args) {
        this.setState(previousState => {
            // scan ENDED!
            return { loading: false };
        });
        console.log(BluetoothService.getDevices());
    }

    handleNewDevice(peripheral) {
        console.log(peripheral);
        Alert.alert("New BLE device", "ID: " + peripheral.id + "\nNAME: " + peripheral.name);
    }
}
```
And this is the service class:

```
import BleManager from 'react-native-ble-manager';
import { Alert, NativeEventEmitter, NativeModules } from 'react-native';

const BleManagerModule = NativeModules.BleManager;

export class BluetoothService {
    static initialize() {
        // inizlalizzazione del modulo del bluetooth
        BleManager.start({ showAlert: true, forceLegacy: true }).then(() => {
            console.log("Module initialized");
        });
    }

    static enableBlth() {
        return BleManager.enableBluetooth();
    }

    static initScan() {
        return BleManager.scan([], 30, true);
    }

    static getDevices() {
        return BleManager.getDiscoveredPeripherals([]);
    }

    static getStatus() {
        BleManager.checkState()
    }
}
```

marcosinigaglia, 2017-09-21:
Hi, try with forceLegacy as false. In the android log did you see something?

HighSoftWare96, 2017-09-21:
With forceLegacy to false nothing changes... For the android log do you mean the JS debugger (like localhost:8081)?

marcosinigaglia, 2017-09-21:
I mean using `adb logcat`.

HighSoftWare96, 2017-09-21:
Ok, sorry, I've not so much experience... What kind of message I've to looking for?

HighSoftWare96, 2017-09-21:
`BluetoothManagerService:Bluetooth is in LE only mode`
I've found this... Maybe it finds out only LE devices?

marcosinigaglia, 2017-09-21:
Filter by "logs" tag.

HighSoftWare96, 2017-09-21:
09-21 13:44:21.068 29933 29992 D logs    : BleManager created
09-21 13:44:22.696 29933 29994 D logs    : start
09-21 13:44:22.700 29933 29994 D logs    : BleManager initialized
09-21 13:44:22.780 29933 29994 D logs    : checkState
09-21 13:44:22.782 29933 29994 D logs    : state:on
09-21 13:45:42.731 29933 29994 D logs    : scan
09-21 13:46:12.923 29933 29994 D logs    : Get discovered peripherals
Nothing more...
I've just tried on a OnePlus 3 device and the application works, while on an another Huawei it doesn't... 
Maybe it's a Huawei problem? I've tried the same thing using an application based on [Polidea/react-native-ble-plx](https://github.com/Polidea/react-native-ble-plx) and making an application with the NativeScript framework but the same problem occurred...

marcosinigaglia, 2017-09-21:
I can't help sorry, there are a lot of article about Huawei P9 ble problem, maybe is a device problem.
Do you have the same problem with other ble apps?

HighSoftWare96, 2017-09-21:
Nope... Besides the built-in system BLE application finds out all the devices...

HighSoftWare96, 2017-09-21:
Already done by this:```

// richiesta permessi localizzazione
        if (Platform.OS === 'android' && Platform.Version >= 23) {
            PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION).then((result) => {
                if (result) {
                    console.log("Permission is OK");
                } else {
                    PermissionsAndroid.requestPermission(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION).then((result) => {
                        if (result) {
                            Alert.alert("User accept");
                        } else {
                            Alert.alert("User refuse");
                        }
                    });
                }
            });
        }
```
right?

HighSoftWare96, 2017-09-21:
I've just tried with a random bluetooth application from the store and the behaviour is the same...

marcosinigaglia, 2017-09-21:
Also in the AndroidManifest.xml . Then check in the Android settings if the app have the permission.

HighSoftWare96, 2017-09-21:
```
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
```
This should be ok... Also on the phone..
Right now I think it's a Huawei problem... 👎

marcosinigaglia, 2017-09-21:
If you want to try you can use this library https://github.com/iDevicesInc/SweetBlue in the scan part of the module.

HighSoftWare96, 2017-09-22:
Ok thanks, but in order to use that library within react-native where I have to put the code since it is Java code?

marcosinigaglia, 2017-09-25:
The android module is in java

closed 2017-10-13 by marcosinigaglia

jiten23, 2018-08-25:
hey I implemented  ble manager. but not scan my bluetooth connected device. I am using my car bluetooth. have any solution?

anwarhamr, 2019-02-27:
I have an app that has been working for months up until this week or last.  I run on multiple devices from android to ios.  Running on a onePlus 3T device it stopped working.  Scans return nothing.  I've since tried on the onePlus 5T which also does not work.  It is running on Samsung S5, S8; the app works as expected.  I've also run it on the Pixel 3, this it also works.  Do you have any idea how I can track this down?  
I was running  I was running ble-mgr 6.2x and have since upgraded to latest with nothing different than what is mentioned above.
I'll be back next week to handle any thoughts or responses.  Thanks in advance.