innoveit/react-native-ble-manager #217: Unable to find any devices. IOS

Sprit3Dan, 2017-09-21:
Hello! Tried to use the plugin, figured how to subscribe to events, that have to be spawned while scanning via `BleManagerDiscoverPeripheral`. Everything went well, though, no events were dispatched. Tried to use your example project from repo, same thing. No exceptions, no logs, just no devices discovered, though through IOS settings I can see few available to connect.

Is there any suggestion where to explore further?

"react-native": "0.48.3",
"react-native-ble-manager": "^6.1.1",
tested on Iphone 7 IOS 11 & Iphone 5s IOS 10.3.3

marcosinigaglia, 2017-09-21:
Did you see peripherals If you use LightBlue app?

Sprit3Dan, 2017-09-21:
@marcosinigaglia, Yeah, LightBlue found devices, though I had to restart my bluetooth on app launch, since app didn't catch that my BT is enabled. Could this be the case and the issue is somewhere here?

Sprit3Dan, 2017-09-21:
Well, it certainly helped. Looks like there's an issue with Bluetooth status on app startup, and it prevents plugin to perform well. Thanks for pointing me out to LightBlue!

marcosinigaglia, 2017-09-21:
Check that you have to call the `start` function before other calls and only one time.

Sprit3Dan, 2017-09-21:
Truth is that I found this issue earlier and developed the bluetooth-service the right way as I understand. Here's the code.
```
import { Observable } from 'rxjs/Rx';
import * as BleManager from 'react-native-ble-manager'
import { NativeEventEmitter, NativeModules } from 'react-native'

import { injectable } from '../';

abstract class BluetoothServiceApi {
	abstract searchDevice(services?: Array<any>, seconds?: number): Observable<any>
}

@injectable()
export class BluetoothService {
	private bluetoothReady: Promise<BluetoothServiceApi>;
	private bleManagerEmitter: NativeEventEmitter;

	private api = {
		searchDevice: this.searchDevice.bind(this),
	};

	constructor() {
		this.bleManagerEmitter = new NativeEventEmitter(NativeModules.BleManager);

		this.bluetoothReady = new Promise((res, rej) => {
		BleManager.start({ showAlerts: true, forceLegacy: true })
			.then(() => {
				res(this.api);
				console.info('bt started');
			});
		});
	}

	public get ready(): Promise<BluetoothServiceApi> {
		return this.bluetoothReady;
	}

	private searchDevice(services: Array<any> = [], seconds: number = 5): Observable<any> {
		let scanEmitter;
		
		BleManager.scan(services, seconds)
			.then(() => console.info('scan started'));
			
		this.bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', (data) => {
			console.log(data);
		});
			
		return new Observable<any>(emitter => scanEmitter = emitter);
		// return Observable.fromEvent(this.bleManagerEmitter, 'BleManagerDiscoverPeripheral');
	}
}
```

and the usage of it.

```
@inject(BluetoothService) btService: BluetoothService;
  
  componentWillMount() {
    this.btService.ready.then(api =>
      api.searchDevice().subscribe(data => console.log(data))
    );
  }
```

Though, it didn't help much

closed 2017-09-22 by Sprit3Dan

Ilario17, 2019-02-28:
@Sprit3Dan how did you solve this issue?