innoveit/react-native-ble-manager #218: iOS does not disconnects after second BleManager.start

stephenjen, 2017-09-26:
### Version

Tell us which versions you are using:

- react-native-ble-manager v 3.3.0
- react-native v 0.43.3
- iOS v 11.0

In my app BLE functionality is spread over two different components, where the first handels the reading/writing while the second handles the pairing.  This is the normal operation sequence:

1.  App starts, no BLE device saved so does nothing
2. Go to pairing component, where it runs BleManager.start and scan, etc.  The end result is the device ID is saved.
3.  Go back to main screen where now a second BleManager.start so we can connect using the saved device ID
4.  In Android it works fine and device and smartphone connects and we're good to go.  But in iOS, for some reason, I cannot connect using BleManager.connect(id, dispatch).  I don't get any errors, it just does not connect.  

I did a test where once the pairing is made and the device connected (BleManager.start already called to do these steps), I call BleManager.start again and then try to connect.  In iOS, the second BleManager.start immediately kicks my device into pairing mode and the reconnect never succeeds. 

What might be causing this to fail in iOS?

Thanks for any pointers!

marcosinigaglia, 2017-09-26:
You have to call `start` only one time.

closed 2017-09-26 by marcosinigaglia

stephenjen, 2017-09-26:
I think this is what you mean - you can only call start one time.  I resolved this by putting all the actions that need to use BLEManager into one action file.