innoveit/react-native-ble-manager #223: Wired crash on Android when App is in the background.

melkayam92, 2017-10-02:
### 6.2.1

Tell us which versions you are using:

- react-native-ble-manager v6.2.1
- react-native v0.42.3
- iOS/Android - Android 7

### Expected behaviour

App shoudn't crash.

### Actual behaviour

App crash, and the BLE Peripheral is locked on the Android device, means i can't reconnect unless i turn off the Peripheral.

Tell us what happens instead

1. Open the App.
2. Scan for devices.
3. Conncet to Device.
4. Disconnect from Device.
5. Moving App to background. (pressing home button)
6. After sometime (20Min), going back to App. (App wasn't killed).
7. Scannig devices. (Got the device)
8. Connect to device - here the App crashes, but the wired thing that it's connect successfully because when i reload the app and try to scan again - i can't find the device (means he already connected to "another" device) and the only way to reconnect is to turn-off the Peripheral. 

I use google analytics so i can only see the lines of the crash: 
  1. NullPointerException (@Peripheral:findReadableCharacteristic:526) {mqt_native_modules}
  2. NullPointerException (@Peripheral:findReadableCharacteristic:548) {mqt_native_modules}

Also, it's not deterministic so its very hard to debug this.. (I only encounter this on release APK, i can't make it happen on debug)

Any idea what can it be?

marcosinigaglia, 2017-10-06:
Hi, if you have this error you read something, is it correct?

melkayam92, 2017-10-07:
Yes, after I connect I read some values... 
It seems like the services object is null, and then I get the NullPointerException. (I'm calling reteriveServices before I read).

For now, i will try to catch the error in the read function in Java side and will pass the rejection to the JS side. 

By the way, you know how to catch the exceptions of the bridge?

closed 2018-01-11 by marcosinigaglia