innoveit/react-native-ble-manager #226: Managing descriptors

Sprit3Dan, 2017-10-06:
Hello, I'm pretty new to BLE and hope you could help me a bit.
Building a RN application, that should connect via bluetooth to some peripherals. The question is,
can I handle the BLE descriptors with this plugin? 

Found this under the device developer guide
`After the BLE connection and App write"0x0002" to a Client Characteristic Configuration
descriptor, Blood Pressure Measurement will be sent on Indication.`

Googled it a little bit, looks like I have to simple write `0x0002` hex to `0x2902` descriptor to allow notification. Not sure why `startNotification` for correct characteristic didn't help a lot. Maybe you have any suggestion for this case?

marcosinigaglia, 2017-10-09:
Hi, you have to use `startNotification`, try to paste here some codes.

Sprit3Dan, 2017-10-09:
okay, here you go. Once a device is discovered this code is run.
```
public init(): void {
  this.btManager.ready
    .then(api => api.connect(this.uuid) //uuid is static for instance of BluetoothDevice
      .then(() => {
        this.services.forEachCharFrom('bloodPressureService')
	  .subscribe(char => api.listen(
	    this.uuid,
	    this.services.getService('bloodPressureService').uuid,
	    char.uuid
	  ));
      })
    );
}
```
where `api.listen` is 
```
private listen(deviceId: string, serviceUid: string, measUid: string): Promise<void> {
  return BleManager.startNotification(deviceId, serviceUid, measUid);
}
```

marcosinigaglia, 2017-10-10:
Sorry @Sprit3Dan , I don't understand your code, what is `forEachCharFrom`?

Sprit3Dan, 2017-10-10:
@marcosinigaglia, that is just a shorthand for iterating over a list of characters from specific service. Don't take that in mind. The only thing that is interesting here is this.
```
api.listen(
  this.uuid, // device UUID
  this.services.getService('bloodPressureService').uuid, // service UUID
  char.uuid // character UUID
));
```

where `api.listen` is defined in my previous comment. The real question is how `BleManager.startNotification` is connected with this statement:

```
After the BLE connection and App write 
"0x0002" to a Client Characteristic Configuration descriptor, 
Blood Pressure Measurement will be sent on Indication.
```

Is it supported with `react-native-ble-manager`, or we have to develop our own BLE plugin for RN?

marcosinigaglia, 2017-10-10:
The indication and the notification is very similar, in iOS for example there isn't a specific method, I think the current module is already ok for the indication.
You have to handle the BleManagerDidUpdateValueForCharacteristic to read the measurement.

Sprit3Dan, 2017-10-10:
ok, i'll investigate further. Thanks! Btw, maybe you have some experience with connection to blood pressure monitors? It would be really handy

closed 2017-12-04 by marcosinigaglia