innoveit/react-native-ble-manager #228: App Crashing on Bluetooth Disconnect

elailai94, 2017-10-10:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.48.4
- iOS v.11.1

### Expected behaviour

The app shouldn't be crashing since we had use NativeEventEmitter as suggested by the examples in the documentation.

### Actual behaviour

Upon disconnection, the following exception is thrown:
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x1c0a6d840> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported

### Steps to reproduce

1. Click on bluetooth unpairing button.
2. The onDisconnect listener method executes and throws an exception.

### Stack trace and console log

2017-10-10 17:01:53.672721-0400 App[1342:358178] BleManager initialized
2017-10-10 17:01:53.682489-0400 App[1342:358178] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x1c0a6d840> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported

marcosinigaglia, 2017-10-11:
Hi, you have to use the `restoreIdentifierKey` option in the start method.

elailai94, 2017-10-11:
Hi, @marcosinigaglia! Thank for your prompt response! As follow-up questions, I just wanted to ask three questions:

1. What is the role of the restore identifier key in the start method?
2. What should I provide as a restore identifier key?
3. In what case is the restore identifier key required and not optional?

Thank you so much for your help once again. I truly appreciate it. 😄

marcosinigaglia, 2017-10-12:
1. Is the way to restore the bluetooth state in iOS when an app go in background
2. https://developer.apple.com/documentation/corebluetooth/cbcentralmanageroptionrestoreidentifierkey?language=objc
3. Maybe is better to set a random UIC to prevent this error.

closed 2017-12-21 by elailai94