innoveit/react-native-ble-manager #230: not same mac address with getBondedPeripherals and BleManagerDiscoverPeripheral listener on android 5

viperasi, 2017-10-12:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.48.3
- Android v.5.1.1

### Problem

Get ble device list with getBondedPeripherals, then connect with mac address, but always error.
Then i use BleManagerDiscoverPeripheral listener, it's worked.
so i watch log and found the mac address is '0F:**03**:15:81:70:AF' from getBondedPeripherals, and address is '0F:**04**:15:81:70:AF' from BleManagerDiscoverPeripheral, and '04' version is correct.

### Code 

getBondedPeripherals
```
BluetoothManager.getBondedPeripherals()
        .then(bondedPeripheralsArray=>{
            console.log('bonded array---->',bondedPeripheralsArray);
            if(bondedPeripheralsArray.length>0){
                // BluetoothManager.connect(bondedPeripheralsArray[0].id) // it's not work
                BluetoothManager.connect('0F:04:15:81:70:AF')  //it's work, address from BleManagerDiscoverPeripheral 
})
```

BleManagerDiscoverPeripheral
```
handleDiscoverPeripheral=(data)=>{
        let id;  
        let macAddress;           
        if(Platform.OS == 'android'){
            macAddress = data.id;
            id = macAddress;
        }
        console.log('macAddress---->', macAddress);         
    }
```

marcosinigaglia, 2017-10-12:
Hi, did you try with other os version?

whitedogg13, 2017-10-12:
Hi @marcosinigaglia this API is only available in Android. Currently there's no reliable approach to get the bonded list in iOS.

@viperasi This API simply calls the corresponding `getBondedDevices` from `BluetoothAdapter` and pack data into json for JavaScript land. It performs no extra logic for mac address.

To see the correctness of this API, you can simply dump the returned bonded devices and compare it with your bluetooth setting UI, and should be correct.

However, by viewing your code, I think there're some points you can consider:
* how can you be sure about the index 0 will be the one you like to connect? There might be multiple bonded devices, and you need a reliable way to confirm which device you want to deal with.
* The `getBondedDevices` API always returns bonded devices, even if these devices might not available right now. The main purpose for this API is to test whether your targeting device is already bonded.

If you still have problem, maybe you can try:
1. remove all bonded device in Android Bluetooth settings
2. bond your target device in Android Bluetooth settings
3. call this API, see whether the returned list is correct (now it should be only ONE device)
4. confirm your device is connectable, and connect to the device using the index 0 device in the bonded list.

Thanks!

viperasi, 2017-10-13:
@whitedogg13
I am sure cause it only one device, and `console.log('bonded array---->',bondedPeripheralsArray);` log is below:
```
bonded array----> [{"id":"0F:03:15:81:70:AF","name":"BF4030"}]
```
and BleManagerDiscoverPeripheral log is this:
```
BleManagerDiscoverPeripheral: {"rssi":-54,"advertising":{"bytes":[2,1,6,3,2,240,24,17,6,242,195,240,174,169,250,21,140,157,73,174,115,113,10,129,231,2,10,4,7,9,66,70,52,48,51,48,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"data":"AgEGAwLwGBEG8sPwrqn6FYydSa5zcQqB5wIKBAcJQkY0MDMwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=","CDVType":"ArrayBuffer"},"id":"0F:04:15:81:70:AF","name":"BF4030"}
```

I think the module has some bug, please check.

whitedogg13, 2017-10-26:
@viperasi Sorry for the late response. Kinda busy these days.

I think the issue you want to point out is the correctness of `getBondedPeripheral` API, and I try to isolate the issue with following steps:
1. Find some reference app which is available to show the bonded device info, such as this one https://play.google.com/store/apps/details?id=com.gerondesign.paireddevicesbluetooth, as you can see these app will show your bonded device info, including BT mac address.
2. Write a RN app and console.log the result of `getBondedPeripheral`
3. Compare the result between 1 and 2.

However per my experiment, the API is returning the correct result. 
Maybe you can also try these steps to check as well.

closed 2018-01-11 by marcosinigaglia