innoveit/react-native-ble-manager #232: BleManagerDisconnectPeripheral is being fired constantly.

RobinBobin, 2017-10-17:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.1.0 / v6.2.4
- react-native v0.48.3
- iOS/Android Android 5.1.1

### Expected behaviour

`BleManagerDisconnectPeripheral` is fired after a timeout during which a connection attempt is made.

### Actual behaviour

This event is emitted once a second or so.

### Steps to reproduce

I know my device MAC-address beforehand. I call `BLE.start()` and then `BLE.connect()`. In my `BleManagerDisconnectPeripheral` handler I call `BLE.connect()` again. My device is off.

For the first 10-15 minutes everything works as expected: `BleManagerDisconnectPeripheral` handler is invoked once approximately every 30 seconds (due to `BLE.connect() in it). But after 10-15 minutes this handler starts being invoked every second or two.

Any help will be greatly appreciated.

marcosinigaglia, 2017-10-17:
Hi @RobinBobin , are you sure that is not a peripheral's problem?

RobinBobin, 2017-10-17:
Hi @marcosinigaglia.

Thanks for answering so fast.

I'm not sure in anything (except I'm having this error) :) .Actually I can't imagine how a peripheral might be the cause. Any ideas?

RobinBobin, 2017-10-17:
I've just noticed the following. When everything is working as expected, logcat outputs

    onConnectionStateChange from 133 to 0 on peripheral:<mac-address>

from your component. And when my handler starts being invoked constantly I get

    onConnectionStateChange from 257 to 0 on peripheral:<mac-address>,

that is, 133 changes to 257. They are some private android codes but maybe that means something.

marcosinigaglia, 2017-10-17:
First, why does the device disconnect the application after 30 seconds? In the other situation the is the connection completed before the disconnection?

RobinBobin, 2017-10-17:
My device is intentionally off. I'm checking the reconnection functionality I implemented. I don't know why the connection timeout is 30 seconds, maybe it's some internal android value. But I see time output in logcat and I see that approximately 30 seconds elapse between my line

    console.log("connecting to...");

and

    console.log("connection failed.");

marcosinigaglia, 2017-10-17:
So you try to connect to a peripheral that is not available...the 30 second's timeout is strange, did you see something in the logcat's output in this interval?

RobinBobin, 2017-10-17:
With

    adb logcat -s -v time ReactNative:V,ReactNativeJS:V,System.err:W,logs:V

I get

    10-17 17:13:24.366 I/ReactNativeJS( 6645): Connecting to 'B0:B4:48:DF:C0:84'.
    10-17 17:13:24.371 D/logs    ( 6645): Connect to: B0:B4:48:DF:C0:84
    10-17 17:13:54.386 D/logs    ( 6645): onConnectionStateChange from 133 to 0 on peripheral:B0:B4:48:DF:C0:84
    10-17 17:13:54.402 I/ReactNativeJS( 6645): No connection to 'B0:B4:48:DF:C0:84'.
    10-17 17:13:54.402 I/ReactNativeJS( 6645): Connecting to 'B0:B4:48:DF:C0:84'.
    10-17 17:13:54.407 D/logs    ( 6645): Peripheral event (BleManagerDisconnectPeripheral):B0:B4:48:DF:C0:84
    10-17 17:13:54.417 D/logs    ( 6645): Connect to: B0:B4:48:DF:C0:84
    ...

The output is the same for the erroneous condition, except 257 is output instead 133 and the timeout is about 1-2 seconds and not 30.

RobinBobin, 2017-10-17:
Eh... sorry to mention. The second parameter in

    onConnectionStateChange(BluetoothGatt gatt, int status, int newState);

is not the old status, but the state of the transition (error, success and so on). So the line

    "onConnectionStateChange from " + status + " to "+ newState + " on peripheral:" + device.getAddress()

might be confusing.

marcosinigaglia, 2017-10-18:
@RobinBobin I have to check this block, if you find something wrong in the `onConnectionStateChange` please make a PR.

closed 2018-01-11 by marcosinigaglia