innoveit/react-native-ble-manager #234: Service and Characteristics Returning Hexadecimal Values instead of 16 bit binary values .

KarthickCSK, 2017-10-25:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

KarthickCSK, 2017-10-25:
For Example :

 BleManager.retrieveServices returned "XXXX" as UUID instead of Proper 16 bit UUID('XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXX'). From which i am unable to read a specific characterstic as it throws an error stating invalid UUID.

vldgeorgiev, 2017-10-26:
@marcosinigaglia I just saw the same issue after upgrading to v6.2.4. The "retrieveServices " method indeed returns only 4 characters from the UUID. I know the UUID of my characteristic, but trying to monitor it or write to it says that the characteristic is not found.
The examples still show passing the UUID as text, so I suppose the format has not changed.
In my case the error is on Android 6.0, RN0.42.
Btw, I checked several BLE versions down to 6.0.0 and they behave the same.

marcosinigaglia, 2017-10-26:
Sorry, I don't have an Android 6 to test

vldgeorgiev, 2017-10-26:
OK. Maybe it happens on other Android versions too. I will see if I can find where the problem comes from.

abikalwa, 2017-10-26:
I too faced the same error . Any suggestions of any alternatives(Other Modules). Because i need it to work on different android versions.

vldgeorgiev, 2017-10-27:
@abikalwa Yesterday I also ran tests with a different module called react-native-ble-plx. It works well, doesn't have problems with the discovery, characteristics, reading/writing, but causes other strange problems while interacting with some actions containing "fetch" calls.
But I guess nothing is perfect :)

KarthickCSK, 2017-10-27:
Thanks @vldgeorgiev . I have tried that as well. Facing some problems while Decoding . When i tried to read the Charcateristic Value. It returns an encoded string  like "UACAAAQAkAE=" . So Did you not face any issue as of such while reading. I considered it to be a Base64 and tried to decode but no luck there as well.

marcosinigaglia, 2017-10-27:
Maybe is a problem in UUIDHelper.uuidToString

KarthickCSK, 2017-10-27:
I hard coded the UUID which i obtained using the ble-plx Module. Even then i get a read failed error.

marcosinigaglia, 2017-10-27:
You need to debug the `findReadableCharacteristic` method and understands what's happen.

abikalwa, 2017-10-28:
@vldgeorgiev  @marcosinigaglia  Even the ble-plx module behaves strangely get some random long strings when i tried to read data from a BLE enabled Wrist band which i am trying to connect and read the data

vldgeorgiev, 2017-10-28:
In my case I don't have problems with the reading and writing with ble-plx. Monitoring a characteristic does return base64 data as expected. But that is with my device, can't say about yours. 
Is it possible that you are trying to monitor a wrong characteristic, or reading it improperly?
The issues that I have with the plx are completely different and maybe specific to my usage. Actually the ble-manager one was more stable, but I tried changing it due to other build issues in the project.

KarthickCSK, 2017-10-28:
@vldgeorgiev  In my case i am trying to scan a  smart wrist band. I am actually unaware of the advertised services and characteristics. Other than the app developed by the wrist band firm . I am facing the same issue even in native android built applications

Anandks1993, 2018-04-23:
I am also getting "XXXX" as uuid

```
{ characteristics:
    [ { properties: { Indicate: 'Indicate' },
        characteristic: '2a05',
        service: '1801' },
      { properties: { Read: 'Read' },
        characteristic: '2a00',
        service: '1800' },
      { properties: { Read: 'Read' },
        characteristic: '2a01',
        service: '1800' },
      { properties: { Read: 'Read' },
        characteristic: '2aa6',
        service: '1800' },
      { descriptors: [ { value: null, uuid: '2902' }, { value: null, uuid: '2901' } ],
        properties: { Notify: 'Notify' },
        characteristic: '2a37',
        service: '180d' },
      { properties: { Read: 'Read' },
        characteristic: '2a38',
        service: '180d' },
      { properties: { Write: 'Write' },
        characteristic: '2a39',
        service: '180d' } ],
   services: [ { uuid: '1801' }, { uuid: '1800' }, { uuid: '180d' } ],
   id: '7B:E7:B4:3E:AF:3D',
   name: 'Default' },
```
Is this is possible to read/write with this data ?
I was using one IOT device which gave 16-bit binary value, so I could able to read/write with that device. But now I am trying to get data from an Android device(version 6.0.1) which returns XXXX as uuid.

closed 2019-06-18 by marcosinigaglia