innoveit/react-native-ble-manager #235: Failed to Read Callback on iOS When Using BleManager.read

mctang719, 2017-10-25:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.46.4
- iOS/Android v.11.0.3

### Expected behaviour

I am trying to use read() to read the response from the client device right after the successful write() sent to the client device.

### Actual behaviour
xcode terminated the process with following error:
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb) 

Thread 1: signal SIGABRT at line: readCallback(@[[NSNull null], [characteristic.value toArray]]);
inside the - (void)peripheral:(CBPeripheral *)peripheral didUpdateValueForCharacteristic:(CBCharacteristic *)characteristic error:(NSError *)error in BleManager class. 

### Steps to reproduce

1. write a 500 bytes packet to the client
2. read from the client for response

### Stack trace and console log

console log: libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb)

marcosinigaglia, 2017-10-26:
Hi, are you sure that you have finish the write before the read?

mctang719, 2017-10-26:
I am not sure, I think the read kicks in earlier than what it suppose to be. 

This is how I am using it:
`onPeripheralClick(peripheral) {
    this.setupManager.connect(peripheral.getId())
        .then(() => {
            ...ble wrtie...
        })
        .then(() => {
            ...ble read...
        })
        .catch((e) => {
            ...error
        }
}`

What is the best practice to read after write is complete?

Thanks

marcosinigaglia, 2017-10-27:
I thinks is wrong, you need something like
```javascript
onPeripheralClick(peripheral) {
	this.setupManager.connect(peripheral.getId()).then(() => {
		ble.write(...).then(() => {
			ble.read(...)
		});
	});
}
```

mctang719, 2017-10-27:
Thanks for the suggestion. I got it working by opening a notification before write, and read from there. Great work on the BLE Manager!

closed 2017-10-27 by mctang719