innoveit/react-native-ble-manager #236: The callback connect() exists in module BleManager, but only one callback may be registered to a function in a native module.

anwarhamr, 2017-10-26:
I'm running an android app connecting to a ble device.  I get the above message when the app is re-loaded/restarted via debugging.  It doesn't always happen but happens enough to be worrisome.  In addition, after I have reloaded my app I get duplicate message that I wouldn't expect to be happening for example multiple messages that a device has been found... 
I have a module that wraps the use of the `BleManager` and it is created/loaded at app startup.  
I'm guessing, probably wrong but guessing, that the native module wireups do not get cleaned up when apps close because there isn't a destructor to do such things.  
Any insight on how these things work would be appreciated.  Thanks

marcosinigaglia, 2017-11-07:
Hi, are you using the hotreload? I think that in some way you init the module twice.

anwarhamr, 2017-11-07:
I am using hot reloading but have also experienced it, not often, when I do a clean start.  But even that is done in quick succession.  
Is there anyway to force cleanup the wire-ups?  I know when hot reloading is being used it is calling my constructor again but the native resources do not get reset/reinitialized.  Any way to make this happen?

marcosinigaglia, 2017-11-08:
I never used the hot reloading or other automation for this reason and I reload the application manually, I don't know if there is a way to prevent this problem.

anwarhamr, 2017-11-09:
Ya, it's a real pain to do it manually all the time.  Since it does not disconnect the client never knows to restart doing what it's supposed to do and my notifications never get wired up... I've tried to getConnected devices but there isn't any, however in reality the phone is still wired up to the client.  Ug.  I hope to have some time to play with this in the next month, for now I just have to get this POC finished.

closed 2018-01-11 by marcosinigaglia